notro/tinydrm #5: Mode Test Failing to Open Device

Tin-Man-Dylan, 2017-07-28:
Currently I am trying to get and ili9488 screen working with 3 wire spi, specifically this one from buydisplay.com (http://www.buydisplay.com/default/lcd-3-5-inch-320x480-tft-display-module-optl-touch-screen-w-breakout-board). To do this I am using a Raspberry Pi Zero W and running Jessie Lite. In order to get this working I have decided to go off of the piscreen module and device tree that are available here. I have made some modifications to the device tree (piscreen2r-overlay.dts) and now it looks like this. From what I have been able to learn, which is mainly from websites like [(kernel.org](https://www.kernel.org/doc/Documentation/devicetree/bindings/spi/spi-bus.txt), and [raspberry pi documentation](https://www.raspberrypi.org/documentation/configuration/device-tree.md). I also put the line "dtoverlay=" and "dtoverlay=piscreen2r" the file "/boot/config.txt" to make it use that overlay. Also "dtparam=spi=on" this line I have uncommented in that file as well. 

For the device tree below I am setting pin 25 to be the reset pin, and pin 22 to be the back light pin. From what I understand looking at the pi zero's default device tree, it isn't necessary to set anything for the chip select(CS) and for the spi pins since these should be the default. Being GPIO10 for MOSI and GPIO11 for SCLK, and for the CS i used GPIO8.
`
/*piscreen2r-overlay.dts*/

/dts-v1/;
/plugin/;

/ {
	compatible = "brcm,bcm2835", "brcm,bcm2708", "brcm,bcm2709";

	fragment@0 {
		target = <&spi0>;
		__overlay__ {
			status = "okay";

			spidev@0{
				status = "disabled";
			};

			spidev@1{
				status = "disabled";
			};
		};
	};

	fragment@1 {
		target = <&gpio>;
		__overlay__ {
			piscreen2_pins: piscreen2_pins {
				brcm,pins = <25 22>;
				brcm,function = <1 1>; /* out out */
			};
		};
	};

	fragment@2 {
		target = <&spi0>;
		__overlay__ {
			/* needed to avoid dtc warning */
			#address-cells = <1>;
			#size-cells = <0>;

			piscreen2: piscreen2@0{
				compatible = "ozzmaker,piscreen2";
				pinctrl-names = "default";
				pinctrl-0 = <&piscreen2_pins>;

				reg = <0>;
				spi-max-frequency = <64000000>;
                spi-3wire;

				rotation = <0>;
				reset-gpios = <&gpio 25 0>;
				backlight = <&backlight>;
			};
		};
	};

	fragment@3 {
		target-path = "/soc";
		__overlay__ {
			backlight: backlight {
				compatible = "gpio-backlight";
				gpios = <&gpio 22 0>;
			};
		};
	};

	__overrides__ {
		speed =		<&piscreen2>,"spi-max-frequency:0";
		rotation =	<&piscreen2>,"rotation:0";
		};
};`

From here to get things up and running I followed the documentation in [raspberry pi developement](https://github.com/notro/tinydrm/wiki/board:-Raspberry-Pi) I ran "uname -a" to get my kernel version which is 4.10.4+. So in that previous link I went through the development instructions following them step by step. The only modification being creating the device tree binary (.dtb) using the device tree (.dts) code that i put above. I did not receive any errors while following this so I went on to testing.

For testing I followed what is on the [development page](https://github.com/notro/tinydrm/wiki/Development) under the mode test section. All of the build commands worked correctly. however when I tried to get the connector id using ./tests/modetest/modetest -M "piscreen" -c it gave me the following error
`
failed to open device 'piscreen': No such file or directory
`
So I am wondering for what reason would the device not have become available in that directory? And what can I do to make it available? Thank you for the help! Also I know some of the specifics may not be necessary to getting the solution I need  but I hope by detailing it might be helpful to other.

notro, 2017-07-30:
You should have a line in the kernel log similar to this:
```
[    x.xxxxxx] piscreen spi0.0: fb1:  frame buffer device
```
Try to put the console on that one: ```con2fbmap 1 1```

Tin-Man-Dylan, 2017-07-31:
Hey Notro,

Thanks for the reply, and also thanks for writing this awesome stuff! The following is just an explanation of what I did.

So you were right about checking the kernel log as it was the problem I was running into. I thought I had been checking it but I guess I forgot. I use the command
`dmesg`
to check it. The problem that was occurring for me was that the "dc pin not selected". So this was due to the device tree overlay that I had modified for what I am doing, I had removed the "dc-gpios", this makes sense as it is something that is indicated to be needed in the device driver "piscreen.c" that I am using. In the future once I get thing working completely I will likely remove it from the driver.

The things that I added into the overlay was changing these to the following
`brcm,pins = <25 0 22>;
brcm,function = <1 1 1>; /* out out */`
And adding this in the piscreen2 section further below
`dc-gpios = <&gpio 0 0>;`

Now all of the modetest commands are working correctly for me, thanks! So the screen is at least turning on and off when I boot the pi and run commands. However things are still not working for me we I will detail in the next comment.

Tin-Man-Dylan, 2017-07-31:
TLDR: Now I am trying to modify the device driver to work for my spi 3-wire, do you have any recommendations for how to approach this in your experience? At the bottom there is a few things I have planned to do.  And included below are the changes I have made so far (just the register initialization)

So now that I have modetest working and the overlay seemingly correct I am hoping to get things displaying. So since I am using SPI-3 wire configuration I figured that some changes would be necessary to the device driver module. Again I decided to go off of the piscreen and just make modifications to it, I think that the current piscreen.c is for doing 4-wire SPI, at least as I can see from the schematic found [here](http://ozzmaker.com/piscreen/PiScreenIIv4.sch.svg)

So being that it is 3-wire the main thing I could think to change was the register initialization variables. Luckily at the Buy Display link previously posted, it included some documentation and sample code. In the sample  code for 3-wire spi in the 9488TP.h file there is a section that has initialization of registers. More info specifically about the registers is also available in the ili9488 data sheet.

So given this information I modified the initialization in piscreen.c to be the same sequence as in the sample code that I found. So it is now as follows:
`	mipi_dbi_hw_reset(mipi);

	mipi_dbi_command(mipi, 0xE0,0x00,0x03,0x09,0x08,0x16,0x0A,0x3F,0x78,0x4C,0x09,0x0A,0x08,0x16,0x1A,0x0F);
	mipi_dbi_command(mipi, 0xE1,0x00,0x16,0x19,0x03,0x0F,0x05,0x32,0x45,0x46,0x04,0x0E,0x0D,0x35,0x37,0x0F);
	

	mipi_dbi_command(mipi, 0xC0,0x17,0x15);
    mipi_dbi_command(mipi, 0xC1,0x41);
	mipi_dbi_command(mipi, 0xC5,0x00,0x12,0x80);
	mipi_dbi_command(mipi, 0x36,0x48);
	mipi_dbi_command(mipi, 0x3A,0x66);
	mipi_dbi_command(mipi, 0xB0,0x80);
	mipi_dbi_command(mipi, 0xB1,0xA0);
	mipi_dbi_command(mipi, 0xB4,0x02);
	mipi_dbi_command(mipi, 0xB6,0x02,0x02);
    mipi_dbi_command(mipi, 0xE9,0x00);
    mipi_dbi_command(mipi, 0xF7,0xA9,0x51,0x2C,0x82);
    mipi_dbi_command(mipi, 0x11);
    msleep(120);
    mipi_dbi_command(mipi, 0x29);`

Now that I had modified this I decided to compile it into the device driver module on the raspberry pi. In order to do this I did the following steps.

1. Removed the existing piscreen files from the tinydrm folder, or just modify the name using the "mv command
2. Now put the modified piscreen.c into the tinydrm folder, I modified it on my computer so I used WinSCP to put it back on the pi
3. Create a file in that folder called "Makefile" and put the following in it:
`
obj-m += piscreen.o

all:
    make -C /lib/modules/$(shell uname -r)/build M=$(PWD) modules
clean:
    make -C /lib/modules/$(shell uname -r)/build M=$(PWD) clean
`
4. Now just used the command "make" to have it build the device  driver, this creates the (.o,.ko,.mod.o, and .mod.c) files

After doing this I rebooted and ran the modetests again and still a blank screen. I also have run this "con2fbmap 1 1" (to output the console on framebuffer 1) but still nothing, though when I do reboot there does seem to be some signal outputting as I observe on my oscilloscope.

Now I realize that there is still more stuff for me to look into, and likely there is more stuff needing to be modified in the device driver. So my plan is to do the following:

-  Double-check all of the initialization commands, by comparing what I now have in there to the ili9488 datasheet
- Look more into tinydrm information on pages like [this](https://www.kernel.org/doc/html/v4.11/gpu/tinydrm.html) and also by looking into what some of the header files provide in terms of functions
- Try to understand more what the ili9488 wants in terms of messages for spi 3-wire

I am wondering if you have any more recommendations on what can be looked into or steps to take to write a good device driver, as you are definitely very skilled at it. Again thanks for your help!

notro, 2017-08-01:
You can sprinkle your code with printk's to track what's going on.
You can also enable drm debug: https://github.com/notro/tinydrm/wiki/Development#debug

notro, 2017-08-01:
It's possible to get the actual content of the buffers before it's sent to spi, but that requires turning on a DEBUG macro, and I don't think I did that on the dev kernel I built.
http://elixir.free-electrons.com/linux/v4.12/source/include/drm/tinydrm/tinydrm-helpers.h#L58

Tin-Man-Dylan, 2017-08-03:
Thanks again for your help Notro! This part is just details on what I did. My next comment has a couple more questions.

So the most helpful thing from your last reply was definitely using that drm debug. I had turned it on before, however I had not realized that the messages from this would display when doing the command "dmesg".

Following this i started figure out a few more things. What I ended up doing was going back to using the default piscreen.c driver and compiling it it, and using the default device tree overlay "piscreen2r-overlay.dts". However what I had realized is that in this folder "/lib/modules/4.10.4+/extra" it still contained the old "piscreen.ko" and it was actually using this. In your Raspberry Pi Developement guide it indicates to remove this one "/lib/modules/$(uname -r)/kernel/drivers/staging/fbtft" (where it gets uname as 4.10.4+ for me) but I guess dor some verisons it is in other folder. I also found that I think the new device module had to be put in that same folder to work (though i may be mistaken and it could have been a different problem all along).

Anyways now that I have it here when I use "./tests/modetest/modetest -M "mi0283qt" -s 25:320x240@RG16" it does display a black screen with a few colored lines.

Tin-Man-Dylan, 2017-08-03:
So my first question is when I do run that "./tests/modetest/modetest -M "piscreen" -s 25:320x240@RG16", what should I be seeing? I searched online and the best I could find was this [article](https://wiki.linaro.org/xinliang/libdrm/modetest)  which indicates the "-s" option should display colour bars. This is what mine look like below. Is your the same when you run it?

[image]

Tin-Man-Dylan, 2017-08-03:
My next question is, you very recently wrote a new page on using Frame Buffer Immaging [fbi](https://github.com/notro/tinydrm/wiki/Tools). This was helpful in realizing why using fbi normally doesn't work, though this doesn't work for me either. was it working for you? I got the same error that you seem to get in that guide

`fbi -d /dev/dri/card0 adapiluv480x320.png
using "DejaVu Sans Mono-16", pixelsize=16.67 file=/usr/share/fonts/truetype/dejavu/DejaVuSansMono.ttf
trying drm: /dev/dri/card0 ...
ioctl VT_GETMODE: Inappropriate ioctl for device
NOTICE: No vt switching available on terminal.
NOTICE: Not started from linux console?  CONFIG_VT=n?
`
I also checked and I don't have a card1, so I used card0. Would this be a problem?

I'm also wondering, since it needs flushing to output properly for this, does this mean that running xserver on fb1 won't work automatically?

notro, 2017-08-03:
modetest should output this image: https://upload.wikimedia.org/wikipedia/commons/6/66/SMPTE_Color_Bars.svg

The fbi I uploaded worked for me. If you turn on drm debug, you should see lines that says the framebuffer is flushed to the display.
card0 is vc4 on my setup:
```
$ sudo cat /sys/kernel/debug/dri/0/name
vc4 dev=soc:gpu unique=soc:gpu
$ sudo cat /sys/kernel/debug/dri/1/name
mi0283qt dev=spi32766.0 unique=spi32766.0
```

xserver: Maybe you have to remove this file: /usr/share/X11/xorg.conf.d/99-fbturbo.conf

closed 2021-07-07 by notro