notro/tinydrm #7: ili9486 support

sergey-suloev, 2017-12-06:
Hi notro

I need to run  the waveshare compatible 3.5inch 480x320  SPI display  with tinyDRM. The display is built on  ili9486. 
Can I have some instructions/guidelines on how to move forward ?
Thanks

notro, 2017-12-07:
ili9486 is a MIPI DBI compatible controller (uses 0x2a, 0x2b and 0x2c for flushing), so you can look at this one: https://patchwork.freedesktop.org/patch/190886/

I have written some info here: https://github.com/notro/tinydrm/wiki/Development
I see that Waveshare has a lot of SPI displays, do you know if they use the same controller in all of them?

It looks like you want to use PRIME for buffer sharing. Currently tinydrm only support physically continous buffers, I don't know what mali does. I'm working on lifting this restriction.

sergey-suloev, 2017-12-07:
I am not sure about other displays as I am only interested in 3.5inch display.
Not sure about PRIME either. Al I know is that Mali needs double-buffering which is not supported by fbtft, that's why I am trying to get into tinydrm. 
Thank you for the link, I am going to get more knowledge of tinydrm by reading source code and see if I can understand how to create a driver.

notro, 2017-12-07:
I have looked at the DT fragment you sent me, and it has regwidth set to 16.
https://github.com/orpaltech/antenna-analyzer-armbian/blob/master/build/userpatches/kernel/sunxi-next/board_orangepipc/sun8i-h3-waveshare35a.dts

So this is one of those displays with a shift register in front of the controller. Argh, I wish I never published that shift register hack, it has just resulted in poor displays.
https://github.com/notro/fbtft/wiki/SPI-interface-circuit

This means that you have to do your own code instead of using [mipi_dbi_command()](http://elixir.free-electrons.com/linux/latest/ident/mipi_dbi_command). That one sends the command as bytes, but you need every byte to be 16-bits except for the pixel data.
Maybe you can get away with making your own mipi->command callback as this driver had to do: https://cgit.freedesktop.org/drm/drm-misc/tree/drivers/gpu/drm/tinydrm/ili9225.c

sergey-suloev, 2017-12-07:
I see. Does it make sense to add such code to tinydrm  along with mipi_dbi_command() as soon as it is a typical case ?

sergey-suloev, 2017-12-07:
About controllers. It seems like each resolution requires a specific controller, i.e. ili9486 is a 480x320 controller.

notro, 2017-12-07:
> Does it make sense to add such code to tinydrm along with mipi_dbi_command() as soon as it is a typical case ?

I don't want non MIPI stuff in mipi-dbi. That's why the ili9225 driver had to do it's own command callback. It used different commands for windows setting and framebuffer writing.

> About controllers. It seems like each resolution requires a specific controller, i.e. ili9486 is a 480x320 controller.

Each controller has a max resolution and usually that's the one used. Many controllers support the same resolution.

I just remembered that I have a driver already that should support such a shift register configuration:
https://github.com/notro/tinydrm/blob/master/piscreen.c

sergey-suloev, 2017-12-07:
i think my display is very much similar to this piscreen

sergey-suloev, 2017-12-08:
hi, I have a question about backlight, which is present in piscreen. It seems like my display does not provide such a pin. Can I safely remove this code ?

	mipi->backlight = tinydrm_of_find_backlight(dev);
	if (IS_ERR(mipi->backlight))
		return PTR_ERR(mipi->backlight);

notro, 2017-12-08:
Yes you can.

If you use [mipi_dbi_pipe_disable()](http://elixir.free-electrons.com/linux/latest/ident/mipi_dbi_pipe_disable) as your drm_simple_display_pipe_funcs->disable callback, you'll get blanking for free.

sergey-suloev, 2017-12-09:
Hi, more questions here
In my system I have sun4i-drm legacy driver which is configured in device tree. Shall it conflict with tinydrm ? 

Is this enough to create a device overlay for a display  or there should be extra nodes for tinydrm itself ?

notro, 2017-12-09:
It shouldn't cause any conflict.
This is the matching piscreen DT overlay: https://github.com/notro/tinydrm/blob/master/rpi-overlays/piscreen-overlay.dts

sergey-suloev, 2017-12-09:
ok, I prepared my own overlay, similar to this
tinydrm and drm components will be loaded dynamically, right ? do I need any specific nodes for tinydrm itself ?

sergey-suloev, 2017-12-10:
Can you explain pls the following line

https://github.com/notro/tinydrm/blob/master/piscreen.c#L58

notro, 2017-12-10:
The SPI bus is big endian and ARM is little endian. If the SPI controller can't do 16-bit natively (for RGB565), tinydrm has to swap the bytes and send it as 8-bit.

That line is from [mipi_dbi_typec3_command()](http://elixir.free-electrons.com/linux/latest/ident/mipi_dbi_typec3_command) and is used there to send RGB565 as 16-bit if the controller can support it, so it's just a left over from me copying that function and not caring about 16-bit which the Raspberry Pi doesn't support on spi0 (which can do dma).

8/16 bit support is detected in [mipi_dbi_spi_init()](http://elixir.free-electrons.com/linux/latest/ident/mipi_dbi_spi_init):
```c
int mipi_dbi_spi_init(struct spi_device *spi, struct mipi_dbi *mipi,
		      struct gpio_desc *dc)
{
...
	if (dc) {
		mipi->command = mipi_dbi_typec3_command;
		mipi->dc = dc;
		if (tinydrm_machine_little_endian() &&
		    !tinydrm_spi_bpw_supported(spi, 16))
			mipi->swap_bytes = true;

```

sergey-suloev, 2017-12-10:
ok , got it

and one more thing: I noticed that sometimes screen isn't fully filled with black color when rotation = 180. 
I haven't noticed the problem with rotation = 0 and rotation = 90. I'm  not sure is this is related to rotation at all.. Just wonder  what do you think it may be caused by?

sergey-suloev, 2017-12-10:
just checked again.. yes, the problem only exists with rotation = 180.
how it looks: about 1/3 of display area in the bottom is not filled black on boot

notro, 2017-12-10:
Command MIPI_DCS_SET_ADDRESS_MODE controls flip/mirror/rotation. The values used for the different rotations are panel specific. In the past I have just tried various values until I found one that was working. I guess you can look in the datasheet and find which bit you need to flip to do the opposite of 90 which you have working.

sergey-suloev, 2017-12-10:
but rotation itself is working fine, i.e. image is rotating correctly

sergey-suloev, 2017-12-10:
just tried Mali, it gives a memory allocation error

notro, 2017-12-10:
mali isn't that the closed source ARM GPU with a kernel shim module?
Are you trying to do rendering on the GPU and scan it out on the SPI display?

sergey-suloev, 2017-12-10:
kernel module is open source, userspace part is closed, unfortunately

sergey-suloev, 2017-12-11:
hi, I noticed that my new driver does not start reliably, it hangs sometimes during system boot.

sergey-suloev, 2017-12-18:
hi, notro,
can I create DT node for tinydrm a way similar to how it is done for sun4i DRM display engine?
Is there any other way to set a specific cma remory region for tinydrm ? 

de: display-engine {
		compatible = "allwinner,sun7i-a20-display-engine";
		allwinner,pipelines = <&fe0>, <&fe1>;
		status = "disabled";
	};

notro, 2017-12-18:
I don't know if you can attach a memory region to a spi child device.
SPI child devices can't do dma, it's the spi controller that does that. tinydrm hacks around that  by pretending that the spi device can do dma in order to allocate memory. 

```c
int mipi_dbi_spi_init(struct spi_device *spi, struct mipi_dbi *mipi,
		      struct gpio_desc *dc)
{
...
	/*
	 * Even though it's not the SPI device that does DMA (the master does),
	 * the dma mask is necessary for the dma_alloc_wc() in
	 * drm_gem_cma_create(). The dma_addr returned will be a physical
	 * adddress which might be different from the bus address, but this is
	 * not a problem since the address will not be used.
	 * The virtual address is used in the transfer and the SPI core
	 * re-maps it on the SPI master device using the DMA streaming API
	 * (spi_map_buf()).
	 */
	if (!dev->coherent_dma_mask) {
		ret = dma_coerce_mask_and_coherent(dev, DMA_BIT_MASK(32));
		if (ret) {
			dev_warn(dev, "Failed to set dma mask %d\n", ret);
			return ret;
		}
	}
```

I'm working on changing the buffers to vmalloc, so this will change in 4.16 or 4.17.

sergey-suloev, 2017-12-19:
ok, not sure I fully understand the change you are going ot make. is it going to address issue I have?
I think my problem that tinydrm is allocating memory at addresses that are not in cma range assigned  and the gpu driver is complaining about that. 
Is there a way to tell tinydrm to use the memory range provided ?
We tried to put  [ memory-region = <&cma>; ] into the spi child node and to apply the following  patch 
https://github.com/orpaltech/antenna-analyzer-armbian/blob/fa32e47905ffcd99584e2ff75aec6a112d5d0337/build/userpatches/kernel/sunxi-next/DRM-driver-for-Waveshare-3.5inch-display.patch#L43
But there is still no effect on the problem.

sergey-suloev, 2018-01-11:
Hi notro,
I tested the original Waveshare 3.5 inch display with my driver  (I tested a chinese "clone" before) and it appears to have white background and black text in console (inverted colors).
Any idea why this may happen ?

notro, 2018-01-18:
Usually wrong colors are due to rgb/bgr ordering or endian (swap bytes) issues.

sergey-suloev, 2018-01-18:
@notro I tried to change rgb/bgr and it didn't make any difference. 
Are you meaning byte swapping which is made here  ?https://github.com/notro/tinydrm/blob/b2e11392baa30df2c1b0b10a3b3171fbc16d8971/piscreen.c#L48

notro, 2018-01-18:
Yeah, but it shows that this isn't likely a byte swapping problem, since the commands comes through ok in the shift register.
Unless the display controller supports byte swapping, which would be a rare thing. Or I could be wrong.

mipi->swap_bytes controls byte swapping in preparation of the framebuffer for transfer. See [mipi_dbi_fb_dirty()](http://elixir.free-electrons.com/linux/v4.15-rc8/ident/mipi_dbi_fb_dirty)

Add ```#define DEBUG``` before the includes in your driver and set [drm_debug](https://github.com/notro/tinydrm/wiki/Development#debug) (DRM_UT_DRIVER) and you will be able to see what's actually going out on the wire from [tinydrm_spi_transfer()](http://elixir.free-electrons.com/linux/v4.15-rc8/ident/tinydrm_spi_transfer).

closed 2018-06-19 by sergey-suloev

akhilpo, 2018-06-19:
@sergey-suloev hi, were you able to make this display work? Do you mind sharing the code?

sergey-suloev, 2018-06-20:
@akhilpo 
You can use this patch as a starting point for you development. It is a working driver.

https://github.com/orpaltech/armlinux/blob/master/patch/kernel/mainline/v4.18-rc1/common/0091-tinydrm-Add-support-for-ILI9486-display-panels.patch

akhilpo, 2018-06-21:
thanks sergey.

alexryndin, 2020-01-27:
@sergey-suloev @notro  Hello guys. I know it's been a long time, but could you please help me to get my ILI9486 based display working with a tinydrm driver?
Unfortunately, I don't know much about driver creation, but @sergey-suloev provided me with [patch for ili9486 support](https://github.com/orpaltech/armlinux/blob/3750784bb96c6d146b5506f6d2b9991f57c2527b/patch/kernel/stable/v5.5/common/0091-tinydrm-Add-support-for-ILI9486-display-panels.patch). I try to get my display working with Raspberry Pi 4, so I patched and compiled [this Linux fork ](https://github.com/raspberrypi/linux/tree/rpi-5.5.y)by raspberry. Kernel compiled well, but then I heavily stuck. With fbtft driver, the display gets initialized and working when I either:
* Put `dtoverlay=piscreen` in /boot/config.txt (So I guess I have piscreen display)
* Do `modprobe fbtft_device name=ili9486`
But using brand new DRM driver neither of above command help - display stays white, uninitialized. How can I debug a new driver? I found out that I could use [modetest tool](https://github.com/notro/tinydrm/wiki/Development#modetest) but I faced the same problem as Tin-Man-Dylan in [this issue](https://github.com/notro/tinydrm/issues/5) - I get `failed to open device 'ili9486': No such file or directory`. I guess that something wrong with a device tree because nothing appears in `dmesg` about ili9486 even when I do `modprobe ili9486` manually, whereas I get something like this:
```
[   54.552040] spidev spi0.0: Deleting spi0.0
[   54.552862] fbtft_device: GPIOS used by 'piscreen':
[   54.552887] fbtft_device: 'reset' = GPIO25
[   54.552907] fbtft_device: 'dc' = GPIO24
[   54.552926] fbtft_device: 'led' = GPIO22
[   54.552966] spidev spi0.1: spidev spi0.1 125000kHz 8 bits mode=0x00
[   54.552994] spi spi0.0: fb_ili9486 spi0.0 32000kHz 8 bits mode=0x00
[   54.574548] fb_ili9486: module is from the staging directory, the quality is unknown, you have been warned.
``` 

when I use a fbtft driver. Fortunately, Sergey also provided me with his old [device tree patch](https://github.com/orpaltech/armlinux/blob/4e80eba461e4ba67d07baa7705838d94ed9169d0/patch/kernel/stable/v5.5/overlays/0005-bcm2835-add-DT-overlays.patch#L964). I just put bcm2835-waveshare35-drm.dts in `arch/arm/boot/dts/overlays` folder, and fixed Makefile to get new DT compiled. It compiled well, but with no sensible result when I put dtoverlay=bcm2835-waveshare35-drm in /boot/config.txt. All the times, I don't get any error in `vcdbg log msg` (raspberry loader log) about overlays, just `Loaded overlay 'overlayname'`

sergey-suloev, 2020-01-27:
@alexryndin  Is there any special reason why you use this kernel tree ? If not I would recommend you switching to mainline kernel ( git://git.kernel.org/pub/scm/linux/kernel/git/stable/linux.git ) as it is very well supported by previous and current bcm2835 maintainers. 
In mainline architecture you will not  use config.txt heavily to configure your board because it is configured by the device tree only (directly or via overlays). Using overlays will require a bootloader though, for example u-boot ( https://gitlab.denx.de/u-boot/u-boot.git ).

alexryndin, 2020-01-27:
@sergey-suloev Thanks. There is not any special reasons for me to use kernel provided by raspberry, so maybe I will also try mainline kernel too, but anyway, how would it help to get drm driver working?

sergey-suloev, 2020-01-27:
@alexryndin  I don't have any guide  on "how to write a tinydrm driver' in my hand, maybe @notro  has something. You will have to get knowledge here and there, read source code and other resources. In general, you will need  
1) driver module (similar to what I have created)
2) device tree for Raspberry, where you will add your nodes for display and touchscreen, as children of the SPI node

sergey-suloev, 2020-01-27:
@alexryndin  as soon as you are new to mainline kernel I would recommend you to start from learning source code and device tree stuff.
for example, for Raspberry Pi 3, you can start from here https://elixir.bootlin.com/linux/v5.5-rc7/source/arch/arm/boot/dts/bcm2837-rpi-3-b.dts

alexryndin, 2020-01-27:
@sergey-suloev thank you, I will definitely dive there :)
As for now, I already have compiled driver module and "should-work" overlay, what tools should I use to debug it? I think it would be a good starting point too.

sergey-suloev, 2020-01-27:
@alexryndin  You can start with the simplest debugging - just print to console from source code. There is a lot of information in internet on this. You might need to declare #define DEBUG in the very top of your source file in order to use dev_dbg() macro
See https://elixir.bootlin.com/linux/v5.5-rc7/source/include/linux/device.h#L1786

notro, 2020-01-27:
FYI: There's a ili9486 driver on the mailinglist: https://lists.freedesktop.org/archives/dri-devel/2020-January/253118.html (there's a version 4 out, but the ML hasn't picked it up yet). 
I will merge it when the Device Tree maintainer has accepted it. This means it will show up in Linux 5.7.

sergey-suloev, 2020-01-27:
@notro thanks for letting this know.  the driver code is here https://lists.freedesktop.org/archives/dri-devel/2020-January/253119.html.

it seems to be very similar to what I have here https://github.com/orpaltech/armlinux/blob/master/patch/kernel/stable/v5.5/common/0091-tinydrm-Add-support-for-ILI9486-display-panels.patch

notro, 2020-01-27:
> it seems to be very similar to what I have

Yeah, all the MIPI DBI compatible displays are almost identical except for the ->enable() function. This one has a quirk with the special ->command() function though.

alexryndin, 2020-01-28:
> FYI: There's a ili9486 driver on the mailinglist
@notro thank you for information. Is that driver module compatible with present piscreen.dts overlay for bcm2835? And what tools should I use to know if it is overlay or driver problem? As for now, there are a few ili9486 compatible overlays already, but they doesn't seem to work for some reason. Sorry if my question looks dumb, as I said above I have very little driver development experience :-)

sergey-suloev, 2020-01-28:
@alexryndin  What overlay should be used really depends on the kernel of choice. 
If you stick with "classical" Rpi kernel (the one you are already using) then you have to learn how to create overlays for it. That shouldn't be too hard but I personally moved away from it long time ago and therefore can't give you any advice on that. Rpi kernel is pretty much different architecture than the mainline kernel.

sergey-suloev, 2020-02-02:
@notro hi, I am trying to make the driver work
https://lists.freedesktop.org/archives/dri-devel/2020-January/253119.html.
and of course it isn't working ( as usual ).

Could you look at my dmesg when you have time, please ?

https://pastebin.com/ntzbssEv

My concern is  line 213 (SPI isn't using DMA, is this a problem for us ?)
and line 271 (hwmode: htotal 858, vtotal 251, vdisplay 240 - does it look correct for you ?)

Some more details:
root@rpi3-aplus-arm64-stretch:~# cat /sys/class/graphics/fb0/virtual_size 
720,480

For you reference, here is the driver code 
https://github.com/orpaltech/armlinux/blob/master/patch/kernel/stable/v5.5/common/0091-tinydrm-Add-support-for-ILI9486-display-panels.patch

and the overlay
https://github.com/orpaltech/armlinux/blob/6009820b167bb546e23eb99f288802778eb4cf0e/patch/kernel/stable/v5.5/overlays/0005-bcm2835-add-DT-overlays.patch#L964

Would be nice to hear any ideas from you. 
Thanks.

notro, 2020-02-02:
At least the driver is loaded:
```
[    3.108610] [drm] Initialized ili9486 1.0.0 20171208 for spi0.0 on minor 1
```
But I can't see waveshare35_enable() in the log so no one is driving it, hence no output.
```con2fbmap 1 1``` should put fbcon on the display.

The vc4 driver is also loaded and is responsible for the other drm log events.

Sprinkling printk's over the driver is a popular way of finding out what actually happens.

As a last resort when it doesn't work, I look at what actually goes over the SPI bus: https://github.com/notro/tinydrm/wiki/Development#spi-trace

sergey-suloev, 2020-02-02:
@notro yes, "enable" isn't there, this is strange..   
how is it possible that drm calls "command" function without "enable" first ?

DRM output you have mentioned is produced by by our display. In my system I can't see HDMI framebuffer, so I have only fb0. This is probably the reason why "con2fbmap 1 1" fails.

root@rpi3-aplus-arm64-stretch:~# ls -l /sys/class/graphics/
total 0
lrwxrwxrwx 1 root root 0 Feb  2 17:47 fb0 -> ../../devices/platform/soc/3f204000.spi/spi_master/spi0/spi0.0/graphics/fb0
lrwxrwxrwx 1 root root 0 Nov  3  2016 fbcon -> ../../devices/virtual/graphics/fbcon

notro, 2020-02-02:
Your previous pastebin dmesg had the ili9486 fbdev framebuffer on fb1 and it had vc4 log entries.

Your new log has it on fb0. So I assume that you don't have enabled the simplefb fbdev driver that loads with the firmware provided framebuffer (would have been fb0).

fbcon automatically (if enabled) attaches to the first fbdev framebuffer, giving a log entry like this:
```
Console: switching to colour frame buffer device 82x26
```
You don't get that so I guess fbcon is not attached to your framebuffer. That would explain why _enable() is not called.

Have you tried modetest? That's what I use for basic testing: https://github.com/notro/tinydrm/wiki/Development#modetest

sergey-suloev, 2020-02-02:
@notro  you're right..
I don't have this line anymore in my log

[    2.824387] vc4-drm soc:gpu: fb0: vc4drmfb frame buffer device

What could be a reason why I don't have  vc4 fb0  anymore ? 
I have vc4 module loaded, btw.

root@rpi3-aplus-arm64-stretch:~# lsmod
Module                  Size  Used by
raspberrypi_cpufreq    16384  0
brcmfmac              241664  0
vc4                   176128  0
brcmutil               20480  1 brcmfmac
drm_kms_helper        212992  1 vc4
cfg80211              679936  1 brcmfmac
syscopyarea            16384  1 drm_kms_helper
sysfillrect            16384  1 drm_kms_helper
rfkill                 36864  4 cfg80211
sysimgblt              16384  1 drm_kms_helper
fb_sys_fops            16384  1 drm_kms_helper
crct10dif_ce           16384  1
drm                   495616  2 drm_kms_helper,vc4
clk_raspberrypi        16384  2
raspberrypi_hwmon      16384  0
bcm2835_rng            16384  0
i2c_bcm2835            16384  0
rng_core               24576  2 bcm2835_rng
pwm_bcm2835            16384  0
bcm2835_thermal        16384  0
bcm2835_dma            24576  0
virt_dma               16384  1 bcm2835_dma
ip_tables              32768  0
x_tables               40960  1 ip_tables
ipv6                  434176  0
nf_defrag_ipv6         24576  1 ipv6

Can we see the problem  from config file https://pastebin.com/szDhH4rT  ?

sergey-suloev, 2020-02-02:
@notro  I think I am missing CONFIG_FB_SIMPLE, correct ?

notro, 2020-02-02:
> I think I am missing CONFIG_FB_SIMPLE, correct ?

This will give you a framebuffer based on what the firmware gives you. When vc4 loads, it will unload simplefb and take its fbdev index.

> What could be a reason why I don't have vc4 fb0 anymore ?
> I have vc4 module loaded, btw.

The module has refcount zero so no users. Since you have the module, either there's a device (from Device Tree) missing or other dependency is missing. I'm not expert on vc4, but it has several subdevices providing pixelvalve, hdmi, I don't remember them all.

But either way, modetest will put a test image on your display: ```modetest -M ili9486 -s <connector id>:480x320```.

sergey-suloev, 2020-02-02:
@notro 
I did modetest

root@rpi3-aplus-arm64-stretch:~# LIBGL_DEBUG=verbose modetest -M "ili9486" -s 31:480x320@RG16 -v
drmOpenDevice: node name is /dev/dri/card0
drmOpenDevice: open result is 3, (OK)
drmOpenDevice: node name is /dev/dri/card0
drmOpenDevice: open result is 3, (OK)
drmGetBusid returned ''
setting mode 480x320-0Hz@RG16 on connectors 31, crtc 34
freq: 3.40Hz
freq: 3.40Hz

root@rpi3-aplus-arm64-stretch:~#

sergey-suloev, 2020-02-03:
I just tried same kernel (same version, same config) with Rpi 3B and vc4 works there: https://pastebin.com/RG2Ty7UM

sergey-suloev, 2020-02-03:
@notro hi, here is an interesting issue

[    4.348568] [drm:drm_fb_helper_generic_probe] surface width(480), height(640) and bpp(16)
[    4.360369] smsc95xx 1-1.1:1.0 eth0: register 'smsc95xx' at usb-3f980000.usb-1.1, smsc95xx USB 2.0 Ethernet, b8:27:eb:27:c4:5d
[    4.362401] [drm:drm_internal_framebuffer_create] bad framebuffer height 640, should be >= 320 && <= 320
[    4.362548] ili9486 spi0.0: [drm:drm_fbdev_client_hotplug] *ERROR* fbdev: Failed to setup generic emulation (ret=-22)
[    4.362606] ili9486 spi0.0: [drm:drm_fbdev_generic_setup] client hotplug ret=-22

notro, 2020-02-03:
Somehow the fbdev emulation picks an illegal resolution.

We need to look at log entries leading up to the ones you posted.

This is the call chain that picks a mode that is passed on to drm_fb_helper_generic_probe() as width/height:

drm_fbdev_generic_setup -> drm_fbdev_client_hotplug -> drm_fb_helper_initial_config -> __drm_fb_helper_initial_config_and_unlock -> drm_client_modeset_probe -> drm_client_target_preferred:
```c
		DRM_DEBUG_KMS("found mode %s\n", modes[i] ? modes[i]->name :
			  "none");
```

sergey-suloev, 2020-02-03:
@notro  thanks for answer, but I just realized why that happens: I had  CONFIG_DRM_FBDEV_OVERALLOC=200 in my config.

sergey-suloev, 2020-02-03:
Here is my updated status, using Rpi 3B, the display is still blank (no console visible)

***************************************

dmesg   https://pastebin.com/CGAL5VyR

***************************************

root@rpi3-b-arm64-stretch:~# ls -l /dev/fb*
crw-rw---- 1 root video 29, 0 Nov  3 17:16 /dev/fb0
crw-rw---- 1 root video 29, 1 Nov  3 17:16 /dev/fb1

***************************************

modetest experiments:

root@rpi3-b-arm64-stretch:~# modetest -M ili9486 -c
Connectors:
id	encoder	status		name		size (mm)	modes	encoders
31	35	connected	(null)-1       	73x49		1	35
  modes:
	name refresh (Hz) hdisp hss hse htot vdisp vss vse vtot)
  480x320 0 480 480 480 480 320 320 320 320 1 flags: ; type: preferred, driver
  props:
	1 EDID:
		flags: immutable blob
		blobs:

		value:
	2 DPMS:
		flags: enum
		enums: On=0 Standby=1 Suspend=2 Off=3
		value: 0
	5 link-status:
		flags: enum
		enums: Good=0 Bad=1
		value: 0
	6 non-desktop:
		flags: immutable range
		values: 0 1
		value: 0
	4 TILE:
		flags: immutable blob
		blobs:

		value:
	20 CRTC_ID:
		flags: object
		value: 34

---------------------------------------------------------

root@rpi3-b-arm64-stretch:~# modetest -M ili9486 -s 31:480x320 -v
setting mode 480x320-0Hz@XR24 on connectors 31, crtc 34
freq: 3.76Hz
freq: 3.76Hz
freq: 3.76Hz
freq: 3.76Hz
freq: 3.76Hz
freq: 3.76Hz
freq: 3.76Hz

****************************************************

notro, 2020-02-03:
The trace looks correct to be, I can see when  modetest runs _enable() and that the 8-bit commands and parameters are sent as 16-bit for this special display.

sergey-suloev, 2020-02-03:
@notro but  there is still  no _enable() ni dmesg.. why ?

notro, 2020-02-03:
> in your example I can see "freq: 31.44Hz" , why do I have so low "freq: 3.76Hz" ?

You have twice as many pixels to transfer and run at 16MHz, I think I run at 48MHz. That by itself is 6 times slower.

> but there is still no _enable() in dmesg.. why ?

It has to run since the trace shows it, unless somehow you're actually using another driver than the one you think you're using.

[12 more replies]