notro/tinydrm #8: tinydrm_lastclose

sergey-suloev, 2018-02-12:
Hi, what is the substitution for tinydrm_lastclose ?

notro, 2018-02-12:
I don't understand the question, can you elaborate.

sergey-suloev, 2018-02-12:
"tinydrm_lastclose" is a function that is missing in latest kernel. what should be used instead ?

notro, 2018-02-12:
Oh sorry I forgot about that change: drm_fb_helper_lastclose should be used.

sergey-suloev, 2018-02-12:
thx for quick response!

do you have some release info what has been added/changed in 4.16 ?

notro, 2018-02-12:
Major changes are reported here: https://github.com/notro/tinydrm/issues/2

Commit log up to and including 4.16: https://github.com/torvalds/linux/commits/master/drivers/gpu/drm/tinydrm
Commit log for 4.17: https://cgit.freedesktop.org/drm/drm-misc/log/drivers/gpu/drm/tinydrm
tinydrm patches are committed to drm-misc.

closed 2018-06-19 by sergey-suloev