notro/tinydrm #10: st7735r: activate PWM backlight

pdp7, 2018-03-23:
I am using  [Adafruit 1.8" TFT LCD](https://www.adafruit.com/products/358) with a BeagleBone Black.  I created a device tree overlay, [BB-LCD-ADAFRUIT-18-SPI1-00A0.dts](https://github.com/beagleboard/bb.org-overlays/blob/master/src/arm/BB-LCD-ADAFRUIT-18-SPI1-00A0.dts), which will load the [tinydrm st7735r driver](https://github.com/torvalds/linux/blob/master/drivers/gpu/drm/tinydrm/st7735r.c) by David Lechner (@dlech).

I ran into a problem when trying to enable PWM backlight control.  I was able to turn on the backlight on by setting `bl_power` to `0` (`FB_BLANK_UNBLACK`) per [comment](https://github.com/beagleboard/bb.org-overlays/pull/68#issuecomment-371281619) by @notro:
```
echo 0 > /sys/devices/platform/backlight_pwm/backlight/backlight_pwm/bl_power
```
NOTE: `FB_BLANK_UNBLACK` is specified in [sysfs-class-backlight](https://www.kernel.org/doc/Documentation/ABI/stable/sysfs-class-backlight)

@notro also [commented](https://github.com/beagleboard/bb.org-overlays/pull/68#issuecomment-375313677) that:
> the tinydrm driver should make sure the backlight turns on when the display pipeline is enabled.

I am creating this issue to discussion a patch to st7735r that would set `bl_power` correctly for PWM backlight.

pdp7, 2018-03-23:
I added `mipi->backlight->props.power = FB_BLANK_UNBLANK` to `st7735r_probe()` and it appears to solve the issue.

**Is this is a good approach?**
```diff
diff --git a/drivers/gpu/drm/tinydrm/st7735r.c b/drivers/gpu/drm/tinydrm/st7735r.c
index 98ff447f40b4..65e6cd4e801d 100644
--- a/drivers/gpu/drm/tinydrm/st7735r.c
+++ b/drivers/gpu/drm/tinydrm/st7735r.c
@@ -171,6 +171,7 @@ static int st7735r_probe(struct spi_device *spi)
        mipi->backlight = tinydrm_of_find_backlight(dev);
        if (IS_ERR(mipi->backlight))
                return PTR_ERR(mipi->backlight);
+       mipi->backlight->props.power = FB_BLANK_UNBLANK;
 
        device_property_read_u32(dev, "rotation", &rotation);
 

```
Related dmesg output with printk()'s for debugging:
```
[   25.908483] tinydrm/st7735r: st7735r_probe(): mipi->backlight->props=574c5eb5
[   25.915658] tinydrm/st7735r: st7735r_probe(): mipi->backlight->props.power=0x4
[   25.922919] tinydrm/st7735r: st7735r_probe(): mipi->backlight->props.power = FB_BLANK_UNBLANK
[   25.931492] tinydrm/st7735r: st7735r_probe(): mipi->backlight->props.power=0x0
[   25.938754] tinydrm/st7735r: st7735r_probe(): device_property_read_u32(dev, "rotation", &rotation)
[   25.947765] tinydrm/st7735r: st7735r_probe(): call mipi_dbi_spi_init()
[   25.954338] tinydrm/st7735r: st7735r_probe(): mipi_dbi_spi_init(): ret=0
[   25.961077] tinydrm/st7735r: st7735r_probe(): call mipi_dbi_init()
```

notro, 2018-03-23:
Did you try the backlight patchset I mentioned in the other thread?
It looks like it should solve your problem. This is how backlight is enabled now (in [linux-next](https://git.kernel.org/pub/scm/linux/kernel/git/next/linux-next.git/tree/include/linux/backlight.h#n133)):
```c
static inline int backlight_enable(struct backlight_device *bd)
{
	if (!bd)
		return 0;

	bd->props.power = FB_BLANK_UNBLANK;
	bd->props.fb_blank = FB_BLANK_UNBLANK;
	bd->props.state &= ~BL_CORE_FBBLANK;

	return backlight_update_status(bd);
}
```
One problem with the backlight subsystem is that is has 3 properties to enable/disable backlight...

closed 2021-07-07 by notro