notro/tinydrm #12: How to add support for 8080 parallel support

srn-skov-dk, 2018-07-24:
The board I have uses a winstar display (https://www.winstar.com.tw/products/graphic-lcd-display-module/lcd-graphics.html)
The display uses something that looks like a 8080 parallel interface and I assume it supports the MIPI command set.
tinydrm does not support parallel interface today - but it looks like a doable task to add this to mipi-dbi.c, adding a new mipi_dbi_spi(...) function, like what exists for SPI already.

But to do so a new platfrom device or something similar is required.
Would adding a "struct dpi_device" do the trick?
A struct dpi_device should include enough functionality to be useful for more than one device, so the combination will make it simpler to add further parallel displays in the future.
We have today auxdisplay/hd44780,c - but this controller supports only characters and not graphics - so I assume they are out of scope.
Any hints what is the best possible way forward to implement support for the 8080 interface?
I hope that the 6800 interface can be included in the same go, but until now I have no HW to test it with so will concentrate on what I can test.

I know the parallel support from a performance point of view will be bad, but the alternative is to use a char device and some proprietary SW - and prefers a more modern approach.
Any hints how to move forward appreciated, so I avoid basing this on the wrong high-level design.

notro, 2018-08-01:
Initially I did consider adding support for an Intel 8080 type bus. Since bit banging such a bus is insanly slow, I looked at adding a new bus type to Linux. The reason for this is that some SoC's have parallel bus support in hw like the RPi and BBB has.
In the end I dropped it because of the time it will take to get something like this approved in mainline. And a parallel bus eats a lot of pins which on most SoC's is better spent on a real video bus like MIPI DPI.

A DMA supported SPI bus is much faster than a bit banged parallel bus, so I left it with that.

I can understand that someone wants to try this for the fun of it.
You can't get this type of display in the same price range with a SPI bus?

I looked at the datasheet for the RA8822 controller this display uses and it's not MIPI DCS compatible (Display Command Set) which uses commands 0x2a, 0x2b and 0x2c to set the window in RAM that is to be updated.

The easiest way driver wise, is finding a display that has a [ST7586](https://www.buydisplay.com/download/ic/ST7586.pdf) monochrome controller for which there is a driver [drivers/gpu/drm/tinydrm/st7586.c](https://elixir.bootlin.com/linux/latest/source/drivers/gpu/drm/tinydrm/st7586.c).

srn-skov-dk, 2018-08-02:
I decided to give it a spin because I have proprietary HW that have a display connected using a lot of GPIO's and I know other have similar needs. Any decent HW design will use SPI or similar as you also acknowledge, this is likely only relevant for old HW designs - and maybe some cheap displays too.
I have implemented a pardata bus and a pardata platform driver. The pardata platform driver creates pardata_devices for each child node found in the DT.
And then I have implemented a wg160160 driver (winstar display linked in previous post) that utilize this and a pardata-dpi lib which based on the tinydrm infrastructure provides the support functions required by the wg160160 driver.
(The pardata-dpi lib has no dependencies on any mipi thing, as you are right that this is not mipi).
I have everything build able and a relatively clean implementation.
When I have brushed up the device bindings and a bit more I will split it in patches and send to dri-devel, lkml - then we will se the feedback.
I plan to add support for a sparkfun display too - as time permits. That would allow me to verify that the abstraction is OK.
The pardatabus may be used for simple char displays too - that is also on my TODO list.

srn-skov-dk, 2018-08-02:
Patchset posted.
See: https://lore.kernel.org/lkml/20180802193909.GA11443@ravnborg.org/

Looking forward for the responses!

closed 2021-07-07 by notro