notro/tinydrm #15: SPI buffers shouldn't be on stack for DMA

andy-shev, 2019-02-19:
If it happens to SPI master that does DMA for all of the transfers tinydrm may fail.
Easy to debug by enabling `CONFIG_DMA_API_DEBUG=y`.

Example (it's only a tip of an iceberg):

    [   23.750467] DMA-API: dw_dmac_pci 0000:00:15.0: device driver maps memory from stack [probable addr=000000001e49185d]
    [   23.750529] WARNING: CPU: 1 PID: 1296 at kernel/dma/debug.c:1161 check_for_stack+0xb7/0x190
    [   23.750533] Modules linked in: mmc_block(+) spi_pxa2xx_platform(+) pwm_lpss_pci pwm_lpss spi_pxa2xx_pci sdhci_pci cqhci intel_mrfld_pwrbtn extcon_intel_mrfld sdhci intel_mrfld_adc led_class mmc_core ili9341 mipi_dbi tinydrm backlight ti_ads7950 industrialio_triggered_buffer kfifo_buf intel_soc_pmic_mrfld hci_uart btbcm
    [   23.750599] CPU: 1 PID: 1296 Comm: modprobe Not tainted 5.0.0-rc7+ #236
    [   23.750605] Hardware name: Intel Corporation Merrifield/BODEGA BAY, BIOS 542 2015.01.21:18.19.48
    [   23.750620] RIP: 0010:check_for_stack+0xb7/0x190
    [   23.750630] Code: 8b 6d 50 4d 85 ed 75 04 4c 8b 6d 10 48 89 ef e8 2f 8b 44 00 48 89 c6 4a 8d 0c 23 4c 89 ea 48 c7 c7 88 d0 82 b4 e8 40 7c f9 ff <0f> 0b 8b 05 79 00 4b 01 85 c0 74 07 5b 5d 41 5c 41 5d c3 8b 05 54
    [   23.750637] RSP: 0000:ffff97bbc0292fa0 EFLAGS: 00010286
    [   23.750646] RAX: 0000000000000000 RBX: ffff97bbc0290000 RCX: 0000000000000006
    [   23.750652] RDX: 0000000000000007 RSI: 0000000000000002 RDI: ffff94b33e115450
    [   23.750658] RBP: ffff94b33c8578b0 R08: 0000000000000002 R09: 00000000000201c0
    [   23.750664] R10: 00000006ecb0ccc6 R11: 0000000000034f38 R12: 000000000000316c
    [   23.750670] R13: ffff94b33c84b250 R14: ffff94b33dedd5a0 R15: 0000000000000001
    [   23.750679] FS:  0000000000000000(0000) GS:ffff94b33e100000(0063) knlGS:00000000f7faf690
    [   23.750686] CS:  0010 DS: 002b ES: 002b CR0: 0000000080050033
    [   23.750691] CR2: 00000000f7f54faf CR3: 000000000722c000 CR4: 00000000001006e0
    [   23.750696] Call Trace:
    [   23.750713]  debug_dma_map_sg+0x100/0x340
    [   23.750727]  ? dma_direct_map_sg+0x3b/0xb0
    [   23.750739]  spi_map_buf+0x25a/0x300
    [   23.750751]  __spi_pump_messages+0x2a4/0x680
    [   23.750762]  __spi_sync+0x1dd/0x1f0
    [   23.750773]  spi_sync+0x26/0x40
    [   23.750790]  mipi_dbi_typec3_command_read+0x14d/0x240 [mipi_dbi]
    [   23.750802]  ? spi_finalize_current_transfer+0x10/0x10
    [   23.750821]  mipi_dbi_typec3_command+0x1bc/0x1d0 [mipi_dbi]
    ...

notro, 2019-02-22:
Thanks Andy, I've sent a patch.
https://patchwork.freedesktop.org/patch/287514/

andy-shev, 2019-02-22:
Seems fixed after ~5 runs. Thanks!

andy-shev, 2019-03-18:
Didn't see this in today's Linux-next. Is it going to be applied soon?

notro, 2019-03-18:
I planned to apply it along side some other patches, but those have stalled so I will apply it later this week.

notro, 2019-03-20:
I've applied it now, should show up in linux-next tomorrow.
https://cgit.freedesktop.org/drm/drm-misc/commit/?id=a89bfc5d9a0732d84b4de311e27133daa0586316

andy-shev, 2019-03-20:
Seems it's on track now, thanks again!

closed 2019-03-20 by andy-shev