notro/tinydrm #17: fix compile on linux 4.19.66 (raspbian buster)

ladyada, 2019-09-14:
hiya notro - if you're interested, i was just starting to compile this and got compilation errors on the most recent raspbian.
https://github.com/ladyada/tinydrm/commit/c23af730cbd8d7bb5ab840d7d8af92045f4b73fc
i can submit a pr if you like :)

notro, 2019-09-14:
Hi ladyada

There are lots of changes in mainline since tinydrm as a concept is now gone. The tinydrm functionality has been moved into core helpers. In 5.4 the folder is renamed to _tiny_ and will contain regular drm drivers that can fit into one sourcefile:
https://cgit.freedesktop.org/drm/drm-misc/tree/drivers/gpu/drm/tiny

I have now updated this repo and rebased my way up to 5.2

ladyada, 2019-09-14:
got it! raspberry pi is still on 4.19 for now, so i made a mini repo just to get me going writing a driver.
the good news is it was working as long as i kept the name 'mi0283qt' and deleted the existing mi'qt driver. however, when i changed the name to a unique name here
https://github.com/ladyada/outoftree/commit/caf67472c20034f2069215b192f9cc347d3df5da
it no longer inserts into the kernel - i get
```
[  588.486707] st7789v_ada: Unknown symbol mipi_dbi_poweron_conditional_reset (err -2)
[  588.486764] st7789v_ada: Unknown symbol tinydrm_fb_dirty (err -2)
[  588.486826] st7789v_ada: Unknown symbol mipi_dbi_command_stackbuf (err -2)
[  588.486867] st7789v_ada: Unknown symbol tinydrm_shutdown (err -2)
[  588.486909] st7789v_ada: Unknown symbol mipi_dbi_debugfs_init (err -2)
[  588.486975] st7789v_ada: Unknown symbol tinydrm_display_pipe_update (err -2)
[  588.487031] st7789v_ada: Unknown symbol mipi_dbi_spi_init (err -2)
[  588.487076] st7789v_ada: Unknown symbol tinydrm_display_pipe_init (err -2)
[  588.487156] st7789v_ada: Unknown symbol mipi_dbi_pipe_disable (err -2)
[  588.487285] st7789v_ada: Unknown symbol mipi_dbi_command_buf (err -2)
[  588.487388] st7789v_ada: Unknown symbol devm_tinydrm_register (err -2)
[  588.487441] st7789v_ada: Unknown symbol tinydrm_merge_clips (err -2)
[  588.487484] st7789v_ada: Unknown symbol tinydrm_gem_cma_free_object (err -2)
[  588.487525] st7789v_ada: Unknown symbol mipi_dbi_buf_copy (err -2)
[  588.487593] st7789v_ada: Unknown symbol tinydrm_gem_cma_prime_import_sg_table (err -2)
[  588.487651] st7789v_ada: Unknown symbol devm_tinydrm_init (err -2)
[  588.487734] st7789v_ada: Unknown symbol mipi_dbi_enable_flush (err -2)
```
i tried linking it into the modules folder with
```
root@devpi:/home/pi# ln -s /home/pi/hello/st7789v_ada.ko  /lib/modules/`uname -r`
root@devpi:/home/pi# ls -l /lib/modules/`uname -r`/*ko
lrwxrwxrwx 1 root root 29 Sep 14 19:00 /lib/modules/4.19.66-v7l+/st7789v_ada.ko -> /home/pi/hello/st7789v_ada.ko
root@devpi:/home/pi# depmod
root@devpi:/home/pi# modprobe st7789v_ada
modprobe: ERROR: could not insert 'st7789v_ada': Unknown symbol in module, or unknown parameter (see dmesg)
```
i feel like theres something minimal im missing cause like i said, it worked for a few days while i got all the init code working. ive looked online for a bit but couldnt find anything helpful.

notro, 2019-09-15:
It works for me. I started with a fresh installation.

```
$ cat /boot/issue.txt
Raspberry Pi reference 2019-07-10
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 175dfb027ffabd4b8d5080097af0e51ed9a4a56c, stage4

$ sudo apt update
$ sudo apt upgrade
$ sudo apt install raspberrypi-kernel-headers
$ sudo reboot

$ uname -a
Linux raspberrypi 4.19.66+ #1253 Thu Aug 15 11:37:30 BST 2019 armv6l GNU/Linux

$ ls -l /usr/src/
total 16
drwxr-xr-x 24 root root 4096 Sep 15 16:27 linux-headers-4.19.66+
drwxr-xr-x 24 root root 4096 Sep 15 16:27 linux-headers-4.19.66-v7+
drwxr-xr-x 24 root root 4096 Sep 15 16:27 linux-headers-4.19.66-v7l+
drwxr-xr-x  3 root root 4096 Jul 10 01:14 sense-hat

pi@raspberrypi:~/work/outoftree $ make -C /lib/modules/$(uname -r)/build/ M=$PWD modules
make: Entering directory '/usr/src/linux-headers-4.19.66+'
  CC [M]  /home/pi/work/outoftree/hello.o
  CC [M]  /home/pi/work/outoftree/st7789v_ada.o
  Building modules, stage 2.
  MODPOST 2 modules
  CC      /home/pi/work/outoftree/hello.mod.o
  LD [M]  /home/pi/work/outoftree/hello.ko
  CC      /home/pi/work/outoftree/st7789v_ada.mod.o
  LD [M]  /home/pi/work/outoftree/st7789v_ada.ko
make: Leaving directory '/usr/src/linux-headers-4.19.66+'

pi@raspberrypi:~/work/outoftree $ sudo make -C /lib/modules/$(uname -r)/build/ M=$PWD modules_install
make: Entering directory '/usr/src/linux-headers-4.19.66+'
  INSTALL /home/pi/work/outoftree/hello.ko
  INSTALL /home/pi/work/outoftree/st7789v_ada.ko
  DEPMOD  4.19.66+
Warning: modules_install: missing 'System.map' file. Skipping depmod.
make: Leaving directory '/usr/src/linux-headers-4.19.66+'

$ sudo mv /lib/modules/4.19.66+/kernel/drivers/gpu/drm/tinydrm/mi0283qt.ko ~/

$ sudo depmod

$ modinfo st7789v_ada
filename:       /lib/modules/4.19.66+/extra/st7789v_ada.ko
license:        GPL
author:         Limor Fried
description:    Sitronix ST7789V Flexible DRM driver
srcversion:     9C3B82408DA2777C70F8D3E
alias:          of:N*T*Cmulti-inno,mi0283qtC*
alias:          of:N*T*Cmulti-inno,mi0283qt
alias:          spi:st7789vada
depends:        drm,tinydrm,mipi-dbi,drm_kms_helper
name:           st7789v_ada
vermagic:       4.19.66+ mod_unload modversions ARMv6 p2v8

$ sudo modprobe st7789v_ada
$ dmesg
[  952.947283] st7789v_ada: loading out-of-tree module taints kernel.

pi@raspberrypi:~/work/outoftree $ dtc -I dts -O dtb -o minipitft114.dtbo minipitft114-overlay.dts
minipitft114.dtbo: Warning (unit_address_vs_reg): /fragment@0/__overlay__/spidev@0: node has a unit name, but no reg property
pi@raspberrypi:~/work/outoftree $ sudo cp minipitft114.dtbo /boot/overlays/

$ sudo nano /boot/config.txt
dtoverlay=minipitft114

$ sudo reboot
$ dmesg
[   24.653945] st7789v_ada: loading out-of-tree module taints kernel.
[   24.714579] st7789vada spi0.0: spi0.0 supply power not found, using dummy regulator
[   24.714764] st7789vada spi0.0: Linked as a consumer to regulator.0
[   24.762462] [drm] Initialized st7789vada 1.0.0 20190914 for spi0.0 on minor 0
[   24.787760] st7789vada spi0.0: fb1: DRM emulated frame buffer device

```

ladyada, 2019-09-15:
yah it works as is right cause i reverted it to have the same `compatible` as the `mi0283qt.ko` 
https://github.com/ladyada/outoftree/blob/master/st7789v_ada.c#L141
but... ideally though i wouldn't have to replace that module, i would have a totally new module that can sit in the same folder - does that make sense? i think its not possible with linux kernels

notro, 2019-09-16:
That works too, I was just adressing your error messages.

```diff
diff --git a/minipitft114-overlay.dts b/minipitft114-overlay.dts
index 49c49a3..d168124 100644
--- a/minipitft114-overlay.dts
+++ b/minipitft114-overlay.dts
@@ -40,7 +40,7 @@
                        #size-cells = <0>;

                        pitft: pitft@0{
-                               compatible = "multi-inno,mi0283qt";
+                               compatible = "adafruit,minipitft114";
                                reg = <0>;
                                pinctrl-names = "default";
                                pinctrl-0 = <&pitft_pins>;
diff --git a/st7789v_ada.c b/st7789v_ada.c
index 926dd6e..e668faa 100644
--- a/st7789v_ada.c
+++ b/st7789v_ada.c
@@ -138,7 +138,7 @@ static struct drm_driver st7789vada_driver = {
 };

 static const struct of_device_id st7789vada_of_match[] = {
-       { .compatible = "multi-inno,mi0283qt" },
+       { .compatible = "adafruit,minipitft114" },
        {},
 };
 MODULE_DEVICE_TABLE(of, st7789vada_of_match);
```

You need to re-run _depmod_ for module autoloading to work, it needs to pick up the new compatible for its database (maybe this was the step you were missing?).

```
$ cat /sys/bus/spi/devices/spi0.0/of_node/compatible
adafruit,minipitft114

$ modinfo st7789v_ada
filename:       /lib/modules/4.19.66+/extra/st7789v_ada.ko
license:        GPL
author:         Limor Fried
description:    Sitronix ST7789V Flexible DRM driver
srcversion:     83ED4BA48053238845CDA2D
alias:          of:N*T*Cadafruit,minipitft114C*
alias:          of:N*T*Cadafruit,minipitft114
alias:          spi:st7789vada
depends:        drm,tinydrm,mipi-dbi,drm_kms_helper
name:           st7789v_ada
vermagic:       4.19.66+ mod_unload modversions ARMv6 p2v8

$ dmesg
[   24.756489] st7789v_ada: loading out-of-tree module taints kernel.
[   24.759703] st7789vada spi0.0: spi0.0 supply power not found, using dummy regulator
[   24.759874] st7789vada spi0.0: Linked as a consumer to regulator.0
[   24.795022] [drm] Initialized st7789vada 1.0.0 20190914 for spi0.0 on minor 0
[   24.806191] st7789vada spi0.0: fb1: DRM emulated frame buffer device

```

You can also drop the suspend/resume code since the Pi doesn't support it. Also the regulator support can go since the Pi is always powered (unless you have a Linux controlled pmic on the display).

```diff
diff --git a/st7789v_ada.c b/st7789v_ada.c
index 926dd6e..dc229da 100644
--- a/st7789v_ada.c
+++ b/st7789v_ada.c
@@ -15,7 +15,6 @@
 #include <linux/gpio/consumer.h>
 #include <linux/module.h>
 #include <linux/property.h>
-#include <linux/regulator/consumer.h>
 #include <linux/spi/spi.h>

 #include <drm/drm_fb_helper.h>
@@ -306,10 +305,6 @@ static int st7789vada_probe(struct spi_device *spi)
                return PTR_ERR(dc);
        }

-       mipi->regulator = devm_regulator_get(dev, "power");
-       if (IS_ERR(mipi->regulator))
-               return PTR_ERR(mipi->regulator);
-
        mipi->backlight = devm_of_find_backlight(dev);
        if (IS_ERR(mipi->backlight))
                return PTR_ERR(mipi->backlight);
@@ -369,32 +364,11 @@ static void st7789vada_shutdown(struct spi_device *spi)
        tinydrm_shutdown(&mipi->tinydrm);
 }

-static int __maybe_unused st7789vada_pm_suspend(struct device *dev)
-{
-       struct mipi_dbi *mipi = dev_get_drvdata(dev);
-
-       return drm_mode_config_helper_suspend(mipi->tinydrm.drm);
-}
-
-static int __maybe_unused st7789vada_pm_resume(struct device *dev)
-{
-       struct mipi_dbi *mipi = dev_get_drvdata(dev);
-
-       drm_mode_config_helper_resume(mipi->tinydrm.drm);
-
-       return 0;
-}
-
-static const struct dev_pm_ops st7789vada_pm_ops = {
-       SET_SYSTEM_SLEEP_PM_OPS(st7789vada_pm_suspend, st7789vada_pm_resume)
-};
-
 static struct spi_driver st7789vada_spi_driver = {
        .driver = {
                .name = "st7789vada",
                .owner = THIS_MODULE,
                .of_match_table = st7789vada_of_match,
-               .pm = &st7789vada_pm_ops,
        },
        .id_table = st7789vada_id,
        .probe = st7789vada_probe,
```

Now the regulator messages are gone:
```
[   24.689762] st7789v_ada: loading out-of-tree module taints kernel.
[   24.745404] [drm] Initialized st7789vada 1.0.0 20190914 for spi0.0 on minor 0
[   24.763413] st7789vada spi0.0: fb1: DRM emulated frame buffer device
```

notro, 2019-09-16:
I realised that something has changed wrt to spi, DT and module autoloading, since the spi id table entry: st7789vada didn't match up with the compatible: minipitft114 and auto loading still worked.
A few years back it was necessary to either have MODULE_ALIAS() or a spi id table in the driver for auto loading to work on the Pi when adding DT devices.

Something has changed either with userspace or the kernel, since now neither is necessary.
This means that you can remove st7789vada_id[] since the driver will only be used on a Device Tree platform. If you want to keep it, change the entry name to the same as the DT entry: minipitft114.

The same driver can support multiple devices and if so there will be multiple entries in the MODULE_DEVICE_TABLE tables. The tables contains device names, not driver/module names.

Here's some of the things I looked at while researching this:
```
This was before removing st7789vada_id:

$ grep st7789v_ada /lib/modules/4.19.66+/modules.alias
alias of:N*T*Cadafruit,minipitft114C* st7789v_ada
alias of:N*T*Cadafruit,minipitft114 st7789v_ada
alias spi:st7789vada st7789v_ada

After removing the spi id table (and running depmod):

pi@raspberrypi:~/work/outoftree $ modinfo st7789v_ada
filename:       /lib/modules/4.19.66+/extra/st7789v_ada.ko
license:        GPL
author:         Limor Fried
description:    Sitronix ST7789V Flexible DRM driver
srcversion:     89E6C47A4A00E58B39566D3
alias:          of:N*T*Cadafruit,minipitft114C*
alias:          of:N*T*Cadafruit,minipitft114
depends:        drm,tinydrm,mipi-dbi,drm_kms_helper
name:           st7789v_ada
vermagic:       4.19.66+ mod_unload modversions ARMv6 p2v8

$ grep st7789v_ada /lib/modules/4.19.66+/modules.alias
alias of:N*T*Cadafruit,minipitft114C* st7789v_ada
alias of:N*T*Cadafruit,minipitft114 st7789v_ada

# Watch udev events while running: sudo dtoverlay minipitft114
$ udevadm monitor  --environment --udev

UDEV  [258.244190] bind     /devices/platform/soc/20204000.spi/spi_master/spi0/spi0.0 (spi)
ACTION=bind
DEVPATH=/devices/platform/soc/20204000.spi/spi_master/spi0/spi0.0
SUBSYSTEM=spi
DRIVER=st7789vada
OF_NAME=pitft
OF_FULLNAME=/soc/spi@7e204000/pitft@0
OF_COMPATIBLE_0=adafruit,minipitft114
OF_COMPATIBLE_N=1
MODALIAS=spi:minipitft114
SEQNUM=1241
USEC_INITIALIZED=257931005

$ cat /sys/bus/spi/devices/spi0.0/uevent
DRIVER=st7789vada
OF_NAME=pitft
OF_FULLNAME=/soc/spi@7e204000/pitft@0
OF_COMPATIBLE_0=adafruit,minipitft114
OF_COMPATIBLE_N=1
MODALIAS=spi:minipitft114

```

I'm adding these refs for my own benefit if I have to look into this again:
- https://github.com/raspberrypi/linux/issues/2989
- [i2c: core: report OF style module alias for devices registered via OF](https://github.com/raspberrypi/linux/commit/af503716ac1444db61d80cb6d17cfe62929c21df)

C4fishfood, 2020-04-24:
This thread seems to be discussing a error similar to one I'm experiencing. I had a mini PI tft running well, but it has now stopped following an update. I'm running Linux raspberrypi 4.19.97+, and have now get the following error when running python scripts using this display:
OSError: /dev/spidev0.0 does not exist

I get the following when running dmesg | grep spi
[   34.268280] st7789vada spi0.0: spi0.0 supply power not found, using dummy regulator
[   34.268433] st7789vada spi0.0: Linked as a consumer to regulator.0
[   34.392030] [drm] Initialized st7789vada 1.0.0 20190914 for spi0.0 on minor 1
[   34.396348] st7789vada spi0.0: fb1: DRM emulated frame buffer device

Can you pleas advise me how to correct this- do i need to do something to load a module in for whatever the st7789vada is?

paul-1, 2020-10-29:
Anyone having luck porting the ST7789v driver to drm/tiny in kernel 5.4?  I have been able to get it to compile clean, and the driver loads without error.  But unable to get anything to show on the framebuffer.

xlla, 2020-12-27:
> Anyone having luck porting the ST7789v driver to drm/tiny in kernel 5.4? I have been able to get it to compile clean, and the driver loads without error. But unable to get anything to show on the framebuffer.

same issue here.

paul-1, 2020-12-28:
Adafuit has an updated driver that works on 5.4 kernels.

https://github.com/adafruit/Raspberry-Pi-Installer-Scripts

closed 2021-07-07 by notro