notro/tinydrm #21: how to use the driver EE0350BT on ili9488 screen?

GreenFeanor, 2020-05-19:
hi, and thank you for all this content. on my side I try to make work an ili9488 screen in DPI via the driver EE0350BT on Raspian Buster (4.19).

info on I left on these:
https://github.com/notro/tinydrm/issues/17
https://github.com/notro/tinydrm/pull/9/commits/690032b23dd235fb82eb0a343388b5a61f3fd292

but there I found nowhere basic of this EE0350BT driver. see technical sheet if necessary: https://drive.google.com/file/d/1z0sa2pZ5k_1co56-s-dZ_UERXpwmQ-ln/view?usp=sharing

notro, 2020-06-01:
I assume you mean DBI and not DPI. That datasheet doesn't contain the controller init sequence that's necessary to make it work. All MIPI DBI compatible controllers are basically the same except for how to setup the controller. Without that you can't make it work. All the mipi_dbi_command() calls in the *_enable() function sets up the controller.

GreenFeanor, 2020-06-02:
No, I mean DPI. there is the full datasheet of the screen: [ili9488_dataheet](https://www.elecrow.com/download/ILI9488%20Data%20Sheet_100.pdf)

I saw some code to init the screen with command and parameters. But I don't understand what pin I have to use for the init. The datasheet is too vague for me about the DPI.

notro, 2020-06-03:
Ok, DPI is not supported by these drivers, it needs a graphics controller.
For instance this is the code that handles DPI on the Raspbery Pi:
https://elixir.bootlin.com/linux/latest/source/drivers/gpu/drm/vc4/vc4_dpi.c

closed 2021-07-07 by notro