notro/tinydrm #23: AT91 sama5d2 spi

alfredoaudi, 2020-07-03:
Hello Notro,

I'm trying to move from a working fbtft_device to a tinydrm without success.
Its look like that it is a spi issues and I can't find a solution.
I'm using kernel 4.19.79 (but I have the same issue with 5.4) in a at91-sama5d27_som1_ek board.
I followed everything from your development wiki page.
I have an 400x240 ili9327 TFT (only initialization string differs) but I'm using yours mi0283qt driver as start example (tried with ili9341 with same result).
I have checked it with a logic analyzer and the hardware are working correct like the kernel log.

Here are some background information:

**at91-sama5d27_som1_ek.dts**
```
flx4: flexcom@fc018000 {
    atmel,flexcom-mode = <ATMEL_FLEXCOM_MODE_SPI>;
    status = "okay";
...
    spi3: spi@400 {
        compatible = "atmel,at91rm9200-spi";
        reg = <0x400 0x200>;
        interrupts = <23 IRQ_TYPE_LEVEL_HIGH 7>;
        dmas = <&dma0 (AT91_XDMAC_DT_MEM_IF(0) | AT91_XDMAC_DT_PER_IF(1) | 
                AT91_XDMAC_DT_PERID(19))>,
            <&dma0 (AT91_XDMAC_DT_MEM_IF(0) | AT91_XDMAC_DT_PER_IF(1) |
                AT91_XDMAC_DT_PERID(20))>;
        dma-names = "tx", "rx";
        clocks = <&flx4_clk>;
        clock-names = "spi_clk";
        pinctrl-names = "default";
        pinctrl-0 = <&pinctrl_mikrobus_spi &pinctrl_mikrobus1_spi_cs 
            &pinctrl_mikrobus2_spi_cs>;
        atmel,fifo-size = <32>;
        status = "okay"; /* Conflict with uart6 and i2c3. */

        #address-cells = <1>;
        #size-cells = <0>;
        display@1{
            compatible = "multi-inno,mi0283qt";
            reg = <1>;
            spi-max-frequency = <32000000>;
            rotation = <0>;
            pinctrl-names = "default";
            pinctrl-0 = <&pinctrl_mikrobus_display>;
            dc-gpios = <&pioA PIN_PD25 GPIO_ACTIVE_HIGH>;
            reset-gpios = <&pioA PIN_PB2 GPIO_ACTIVE_HIGH>;
            //backlight = <&backlight>;
            status = "okay";
        };
    };
```

**kernel log**
```
atmel_spi f8000000.spi: DMA TX channel not available, SPI unable to use DMA
atmel_spi f8000000.spi: Atmel SPI Controller using PIO only
atmel_spi f8000000.spi: Using FIFO (16 data)
atmel_spi f8000000.spi: Atmel SPI Controller version 0x311 at 0xf8000000 (irq 31)
atmel_spi fc018400.spi: Using dma0chan0 (tx) and dma0chan1 (rx) for DMA transfers
atmel_spi fc018400.spi: Using FIFO (16 data)
atmel_spi fc018400.spi: Atmel SPI Controller version 0x311 at 0xfc018400 (irq 179)
...
mi0283qt spi1.1: spi1.1 supply power not found, using dummy regulator
mi0283qt spi1.1: Linked as a consumer to regulator.0
[drm:0xbf00e904] SPI speed: 32MHz
[drm:0xbf00e5ac] preferred_depth=16, rotation = 0
[drm:0xc03f6248]
[drm:0xc03f6248]
[drm:0xc03f62c8] new minor registered 0
[drm:0xc03f7810] adding "Virtual-1" to sysfs
[drm:0xc03f7668] generating hotplug event
[drm] Initialized mi0283qt 1.0.0 20160614 for spi1.1 on minor 0
[drm:0xc040c2a0] OBJ ID: 28 (2)
[drm:0xc03ee668]
[drm:0xc03dfac4] [CONNECTOR:28:Virtual-1]
[drm:0xc03dfcd0] [CONNECTOR:28:Virtual-1] probed modes :
[drm:0xc03faf28] Modeline 32:"320x240" 0 1 320 320 320 320 240 240 240 240 0x48 0x0
[drm:0xc03ee7e8] connector 28 enabled? yes
[drm:0xc03eed04] looking for cmdline mode on connector 28
[drm:0xc03eef78] looking for preferred mode on connector 28 0
[drm:0xc03eed2c] found mode 320x240
[drm:0xc03ef098] picking CRTCs for 320x240 config
[drm:0xc03ee9ec] desired mode 320x240 set on crtc 30 (0,0)
[drm:0xc040c2a0] OBJ ID: 28 (2)
[drm:0xc03ee264] surface width(320), height(240) and bpp(16)
[drm:0xc0408e40] [FB:34]
[drm:0xc040c240] OBJ ID: 34 (2)
mi0283qt spi1.1: fb0: DRM emulated frame buffer device
[drm:0xc03f6b98]
[drm:0xc03f381c] pid = 224, minor = 0
[drm:0xc03f391c]
[drm:0xc03f513c] pid=224, dev=0xe200, auth=1, DRM_IOCTL_VERSION
[drm:0xc03f513c] pid=224, dev=0xe200, auth=1, DRM_IOCTL_VERSION
[drm:0xc03f3a4c] open_count = 1
[drm:0xc03f336c] pid = 224, device = 0xe200, open_count = 1
[drm:0xc03f3968]
[drm:0xc03f3990] driver lastclose completed
[drm:0xc040c2a0] OBJ ID: 34 (1)
[drm:0xc040c2a0] OBJ ID: 28 (3)
[drm:0xc040c2a0] OBJ ID: 28 (4)
[drm:0xbf013064]
[drm:0xbf00f2d0] cmd=0a, par=08
[drm:0xbf00f4a0] cmd=01
[drm:0xbf00f4a0] cmd=28
[drm:0xbf00f488] cmd=cf, par=00 83 30
[drm:0xbf00f488] cmd=ed, par=64 03 12 81
[drm:0xbf00f488] cmd=e8, par=85 01 79
[drm:0xbf00f488] cmd=cb, par=39 2c 00 34 02
[drm:0xbf00f488] cmd=f7, par=20
[drm:0xbf00f488] cmd=ea, par=00 00
[drm:0xbf00f488] cmd=c0, par=26
[drm:0xbf00f488] cmd=c1, par=11
[drm:0xbf00f488] cmd=c5, par=35 3e
[drm:0xbf00f488] cmd=c7, par=be
[drm:0xbf00f488] cmd=3a, par=05
[drm:0xbf00f488] cmd=b1, par=00 1b
[drm:0xbf00f488] cmd=f2, par=08
[drm:0xbf00f488] cmd=26, par=01
[drm:0xbf00f488] cmd=e0, par=1f 1a 18 0a 0f 06 45 87 32 0a 07 02 07 05 00
[drm:0xbf00f488] cmd=e1, par=00 25 27 05 10 09 3a 78 4d 05 18 0d 38 3a 1f
[drm:0xbf00f488] cmd=b7, par=07
[drm:0xbf00f488] cmd=b6, par=0a 82 27 00
[drm:0xbf00f4a0] cmd=11
[drm:0xbf00f4a0] cmd=29
[drm:0xbf00f488] cmd=36, par=e8
[drm:0xbf00e67c] Flushing [FB:34] x1=0, x2=320, y1=0, y2=240
[drm:0xbf00f488] cmd=2a, par=00 00 01 3f
[drm:0xbf00f488] cmd=2b, par=00 00 00 ef
[drm:0xbf00f3c4] cmd=2c, len=153600
spi_master spi1: failed to transfer one message from queue
spi_master spi1: failed to transfer one message from queue
spi_master spi1: failed to transfer one message from queue
[drm:0xc040c240] OBJ ID: 28 (5)
mi0283qt spi1.1: [drm:0xc04167a0] fbdev: ret=0
[drm:0xc03f6b98]
[drm:0xc03f6b98]
```
Please show me the way.
Thanks a lot in advance.

notro, 2020-07-07:
It's the pixel transfer that fails in your case, the framebuffer is split into 3 SPI messages which all fail:
```
[drm:0xbf00f3c4] cmd=2c, len=153600
spi_master spi1: failed to transfer one message from queue
spi_master spi1: failed to transfer one message from queue
spi_master spi1: failed to transfer one message from queue
```

Too bad the error message doesn't print the error code.

This is the call chain in your case:

[mipi_dbi_spi_transfer](https://elixir.bootlin.com/linux/latest/C/ident/mipi_dbi_spi_transfer) -> [...] -> [__spi_pump_messages](https://elixir.bootlin.com/linux/latest/A/ident/__spi_pump_messages) -> [atmel_spi_transfer_one_message](https://elixir.bootlin.com/linux/latest/C/ident/atmel_spi_transfer_one_message) -> [atmel_spi_one_transfer](https://elixir.bootlin.com/linux/latest/C/ident/atmel_spi_one_transfer)

Looks like you need to build your own kernel and add some printk's to find out what's going on.

You could also try and disable DMA on the SPI controller to see if that works, it will be slow but will help narrowing down the problem area.

alfredoaudi, 2020-07-14:
Thanks for your guidance.
In order to follow your call chain I've checkout kernel 5.4.41. So sorry but the log is a bit different from the previous one.
I have turn on the pertinent debug messages (hope so).
```
[drm:0xc046b48c] cmd=2b, par=00 00 00 ef
mi0283qt spi1.1: new message d2b7ca0b submitted for spi1.1
mi0283qt spi1.1: activate NPCS, mr 00010031
atmel_spi fc018400.spi:   start fifo xfer 711cb1e6: len 1 tx 24f5f56c rx bc400fbc bitpw 8
mi0283qt spi1.1:   xfer 711cb1e6: len 1 tx 24f5f56c/0x00000000 rx bc400fbc/0x00000000
mi0283qt spi1.1: DEactivate NPCS, mr 000f0031
mi0283qt spi1.1: new message d2b7ca0b submitted for spi1.1
mi0283qt spi1.1: activate NPCS, mr 00010031
atmel_spi fc018400.spi:   start fifo xfer 711cb1e6: len 4 tx 1392362b rx bc400fbc bitpw 8
mi0283qt spi1.1:   xfer 711cb1e6: len 4 tx 1392362b/0x00000000 rx bc400fbc/0x00000000
mi0283qt spi1.1: DEactivate NPCS, mr 000f0031
[drm:0xc046b3cc] cmd=2c, len=153600
mi0283qt spi1.1: new message 2080fc6f submitted for spi1.1
mi0283qt spi1.1: activate NPCS, mr 00010031
atmel_spi fc018400.spi:   start fifo xfer c09feddd: len 1 tx 1392362b rx bc400fbc bitpw 8
mi0283qt spi1.1:   xfer c09feddd: len 1 tx 1392362b/0x00000000 rx bc400fbc/0x00000000
mi0283qt spi1.1: DEactivate NPCS, mr 000f0031
mi0283qt spi1.1: new message 2080fc6f submitted for spi1.1
mi0283qt spi1.1: activate NPCS, mr 00010031
mi0283qt spi1.1: you can't yet change bits_per_word in transfers
mi0283qt spi1.1: DEactivate NPCS, mr 000f0031
spi_master spi1: failed to transfer one message from queue
[drm:0xc0457d8c] OBJ ID: 31 (4)
mi0283qt spi1.1: [drm:0xc04638f4] fbdev: ret=0
[drm:0xc0443b58]
```
It's look like [atmel_spi_one_transfer](https://elixir.bootlin.com/linux/latest/source/drivers/spi/spi-atmel.c#L1280) is not compatible with drm code?

notro, 2020-07-19:
Yeah you're right, drm_mipi_dbi uses 8-bit for setup commands and switches to 16-bit for pixel transfers and spi-atmel doesn't support that:
```
mi0283qt spi1.1: you can't yet change bits_per_word in transfers
```
If you build your own kernel you have two ways to work around this:

You can force spi-atmel to announce only 8-bit support:
```diff
 static int atmel_spi_probe(struct platform_device *pdev)
 {
+	master->bits_per_word_mask = SPI_BPW_RANGE_MASK(8, 16);
-	master->bits_per_word_mask = SPI_BPW_MASK(8);
 }
```

Or you can trick drm_mipi_dbi into believing that the spi driver only supports 8-bit by unconditionally setting swap_bytes (I think this will work, but I might be wrong):
```diff
 int mipi_dbi_spi_init(struct spi_device *spi, struct mipi_dbi *dbi,
 		      struct gpio_desc *dc)
 {
-		if (mipi_dbi_machine_little_endian() && !spi_is_bpw_supported(spi, 16))
 			dbi->swap_bytes = true;
 }
```

alfredoaudi, 2020-07-21:
I have just test it and both of your solutions worked!
For me the first one makes more sense since it is a spi-atmel driver problem and not a drm_mipi_dbi one.
Thanks a lot!!!

closed 2020-07-21 by alfredoaudi