notro/tinydrm #24: how to use ST7789 Overlay & Driver on raspberry pi 4.

xlla, 2020-12-27:
I am using pi4 with aarch64 OS, attached a 1.3inch lcd HAT using st7789vw controller.
It works fine on kernel 4.19 by fbtft and flexfb.
After migrate to kernel 5.4, it stop work.

I have to comment out all fbtft & flexfb configurations, pull and compile tinydrm, deploy kernel module and load it on boot, modify st7789vw dts to adopt my lcd,
compile and load it in config.txt. 

After those instructions, I still can't make it work.

<pre>
rpi4:~$ lsmod |grep st77
st7789vw               16384  1
drm_mipi_dbi           28672  1 st7789vw
backlight              20480  2 st7789vw,gpio_backlight

rpi4:~$ dmesg |grep st7
[    2.592372] st7789vw: loading out-of-tree module taints kernel.
[    3.842403] st7789vw spi0.0: fb1: ST7789VWdrmfb frame buffer device

rpi4:~$  DISPLAY=:0 xrandr --listproviders
Providers: number : 2
Provider 0: id: 0x41 cap: 0xf, Source Output, Sink Output, Source Offload, Sink Offload crtcs: 1 outputs: 1 associated providers: 0 name:modesetting
Provider 1: id: 0x74 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 0 name:modesetting

</pre>

the lcd shows nothing while execute
<pre>
rpi4:~$ cat /dev/random > /dev/fb1
cat: write error: No space left on device
</pre>

directfb can't work with it
<pre>
(*) Direct/Memcpy: Using Generic 64bit memcpy()
(*) Direct/Thread: Started 'Fusion Dispatch' (1345) [MESSAGING - OTHER/0] <8388608>...
(*) DirectFB/FBDev: Found 'ST7789VWdrmfb' (ID 0) with frame buffer at 0x00000000, 112k (MMIO 0x00000000, 0k)
(*) DirectFB/Graphics: Generic Software Rasterizer 0.7 (directfb.org)
(*) DirectFB/Core/WM: Default 0.3 (directfb.org)
(*) Direct/Thread: Started 'Genefx' (1346) [DEFAULT - OTHER/0] <8388608>...
(*) FBDev/Mode: Setting 240x240 RGB16
(*) FBDev/Mode: Switched to 240x240 (virtual 240x240) at 16 bit (RGB16), pitch 480
(!) DirectFB/FBDev: Could not set gamma ramp    --> Device or resource busy
<pre>

mesa egl can't work with it , and mesa failure to load an driver ST7789VM
<pre>
rpi4:~$ DISPLAY=:0 eglinfo 
EGL client extensions string:
    EGL_EXT_client_extensions EGL_EXT_device_base
    EGL_EXT_device_enumeration EGL_EXT_device_query EGL_EXT_platform_base
    EGL_KHR_client_get_all_proc_addresses EGL_KHR_debug
    EGL_EXT_platform_device EGL_EXT_platform_x11 EGL_KHR_platform_x11
    EGL_MESA_platform_gbm EGL_KHR_platform_gbm
    EGL_MESA_platform_surfaceless

GBM platform:
MESA-LOADER: failed to open ST7789VW: /usr/lib/dri/ST7789VW_dri.so: cannot open shared object file: No such file or directory (search paths /usr/lib/dri)
failed to load driver: ST7789VW
Mesa: The provided filesystem timestamp for the cache is bogus! Disabling On-disk cache.
EGL API version: 1.4
EGL vendor string: Mesa Project
EGL version string: 1.4
EGL client APIs: OpenGL OpenGL_ES 
</pre>

xlla, 2020-12-27:
I have a X11 session running on hdmi display, it seems all fb will be occupied.
After stop X, the error message changed, 
<pre>
(*) Direct/Memcpy: Using Generic 64bit memcpy()
(*) Direct/Thread: Started 'Fusion Dispatch' (1334) [MESSAGING - OTHER/0] <8388608>...
(*) DirectFB/FBDev: Found 'ST7789VWdrmfb' (ID 0) with frame buffer at 0x00000000, 112k (MMIO 0x00000000, 0k)
(*) DirectFB/Graphics: Generic Software Rasterizer 0.7 (directfb.org)
(*) DirectFB/Core/WM: Default 0.3 (directfb.org)
(*) Direct/Thread: Started 'Genefx' (1335) [DEFAULT - OTHER/0] <8388608>...
(*) FBDev/Mode: Setting 240x240 RGB16
(*) FBDev/Mode: Switched to 240x240 (virtual 240x240) at 16 bit (RGB16), pitch 480
(!) DirectFB/FBDev: Could not set gamma ramp    --> Invalid argument
</pre>

modetest runs fine, but no image on lcd.
<pre>
rpi4:~/git$  modetest -M ST7789VW -s 31:#0@RG16 -v
setting mode 240x240-0.02Hz on connectors 31, crtc 34
failed to set gamma: Function not implemented
freq: 16.57Hz
freq: 16.54Hz
freq: 16.55Hz
freq: 16.54Hz
freq: 16.55Hz
^C

</pre>

notro, 2021-01-02:
cc @EliasKotlyar (who made the driver)

closed 2021-07-07 by notro