notro/tinydrm #25: ili9341 on linux 5.9.16 not working properly

iamdrq, 2020-12-29:
Many thanks for tinydrm.
I found the tinydrm lately,I moved fbtft to tinydrm excitedly! but it is not working properly,that look this:

[image]

I don't konw what happed,and I did  some testing:

> root@orangepi4:/boot/overlay-user# modetest -M "ili9341" -c
```
Connectors:
id	encoder	status		name		size (mm)	modes	encoders
31	35	connected	(null)-1       	49x37		1	35
  modes:
	index name refresh (Hz) hdisp hss hse htot vdisp vss vse vtot)
  #0 320x240 0.01 320 320 320 320 240 240 240 240 1 flags: ; type: preferred, driver
  props:
	1 EDID:
		flags: immutable blob
		blobs:
		value:
	2 DPMS:
		flags: enum
		enums: On=0 Standby=1 Suspend=2 Off=3
		value: 0
	5 link-status:
		flags: enum
		enums: Good=0 Bad=1
		value: 0
	6 non-desktop:
		flags: immutable range
		values: 0 1
		value: 0
	4 TILE:
		flags: immutable blob
		blobs:
		value:
	20 CRTC_ID:
		flags: object
		value: 34
```
> root@orangepi4:/boot/overlay-user# modetest -M "ili9341" -s 31:320x240@RG16 -v
```
setting mode 320x240-0.01Hz@RG16 on connectors 31, crtc 34
failed to set gamma: Function not implemented
freq: 1958.35Hz
freq: 1970.31Hz
freq: 1531.32Hz
freq: 1545.03Hz
freq: 2814.94Hz
freq: 4297.66Hz
freq: 4304.13Hz
```
> root@orangepi4:/boot/overlay-user# dmesg |grep spi
```
[    4.737590] [drm] Initialized ili9341 1.0.0 20180514 for spi1.0 on minor 0
[    4.743904] ads7846 spi1.1: supply vcc not found, using dummy regulator
[    4.744910] ili9341 spi1.0: [drm] *ERROR* Failed to update display -22
[    4.746782] ads7846 spi1.1: touchscreen, irq 121
[    4.763146] input: ADS7846 Touchscreen as /devices/platform/ff1d0000.spi/spi_master/spi1/spi1.1/input/input7
[    5.116143] ili9341 spi1.0: [drm] fb0: ili9341drmfb frame buffer device
```
> root@orangepi4:/boot/overlay-user# dmesg |grep drm
```
[    4.149572] systemd[1]: Condition check resulted in Load Kernel Module drm being skipped.
[    4.737590] [drm] Initialized ili9341 1.0.0 20180514 for spi1.0 on minor 0
[    4.744910] ili9341 spi1.0: [drm] *ERROR* Failed to update display -22
[    4.775493] panfrost ff9a0000.gpu: [drm:panfrost_devfreq_init [panfrost]] Failed to register cooling device
[    4.794188] [drm] Initialized panfrost 1.1.0 20180908 for ff9a0000.gpu on minor 1
[    4.861633] rockchip-drm display-subsystem: bound ff8f0000.vop (ops vop_component_ops [rockchipdrm])
[    4.861711] [drm] unsupported AFBC format[3231564e]
[    4.866473] rockchip-drm display-subsystem: bound ff900000.vop (ops vop_component_ops [rockchipdrm])
[    4.866656] rockchip-drm display-subsystem: bound fec00000.dp (ops cdn_dp_component_ops [rockchipdrm])
[    4.868390] cdn-dp fec00000.dp: [drm:cdn_dp_pd_event_work [rockchipdrm]] Not connected. Disabling cdn
[    4.874237] rockchip-drm display-subsystem: bound ff940000.hdmi (ops dw_hdmi_rockchip_ops [rockchipdrm])
[    4.882608] rockchip-drm display-subsystem: [drm] Cannot find any crtc or sizes
[    4.882949] [drm] Initialized rockchip 1.0.0 20140818 for display-subsystem on minor 2
[    4.900521] cdn-dp fec00000.dp: [drm:cdn_dp_pd_event_work [rockchipdrm]] Not connected. Disabling cdn
[    5.116143] ili9341 spi1.0: [drm] fb0: ili9341drmfb frame buffer device
[    5.898522] rockchip-drm display-subsystem: [drm] Cannot find any crtc or sizes
```

I noticed " ili9341 spi1.0: [drm] *ERROR* Failed to update display -22" in dmesg,is problem?

notro, 2020-12-29:
> I noticed " ili9341 spi1.0: [drm] ERROR Failed to update display -22" in dmesg,is problem?

Probably. That message comes from [mipi_dbi_fb_dirty()](https://elixir.bootlin.com/linux/v5.9/C/ident/mipi_dbi_fb_dirty) and is only printed on the first error.
Most likely the error comes from the SPI subsystem.
It's been a while since I last looked at it, but I assume the tracing framework will let you know about failed SPI transfers. See https://github.com/notro/tinydrm/wiki/Development#spi-trace

iamdrq, 2020-12-31:
Yes,you are right,I found this:
[mipi_dbi_spi_transfer](https://github.com/torvalds/linux/blame/f6e1ea19649216156576aeafa784e3b4cee45549/drivers/gpu/drm/drm_mipi_dbi.c#L1131)
```
int mipi_dbi_spi_transfer(struct spi_device *spi, u32 speed_hz,
			  u8 bpw, const void *buf, size_t len)
{
	size_t max_chunk = spi_max_transfer_size(spi);
```
That maybe max_chunk  is too big or spi_max_transfer_size got incorrect  ,so I adjust the value(or have a compatible way get the  spi's max_chunk ):

> max_chunk = 10000

It worked ! At least,found the reason,many thank your reply.

closed 2020-12-31 by iamdrq