notro/tinydrm #27: Second display not getting refreshed properly under X

yam001, 2021-04-05:
I have a dual display setup (using ili9341 screens) working properly using fbtft drivers, but when I move them to use drm drivers, the second display only gets refreshed if I mouse over it, otherwise its not getting refreshed. If I use 1 fbtft driver and 1 drm driver, they both get refreshed properly. This happens only when both of them are using drm drivers. Does not matter if I use two same drm drivers or two different drm drivers. I have been pulling my hair out to try to figure this out. I was wondering if you knew of anything in the driver, drm, or X that would cause this to happen and any recommendation on what I should try? Thanks.

yam001, 2021-04-05:
I was using the latest kernel, 5.11.11.

notro, 2021-04-11:
I haven't heard of this problem before.

I don't know the Xorg internals, but I do know that the DRM ioctl's go through the [modesetting driver](https://gitlab.freedesktop.org/xorg/xserver/-/blob/master/hw/xfree86/drivers/modesetting).

So looking at the log:
```
$ cat /var/log/Xorg.0.log | grep -i damage
[   466.830] (II) Initializing extension DAMAGE
[   466.870] (II) modeset(0): Damage tracking initialized
```

And then looking in [modesetting/driver.c](https://gitlab.freedesktop.org/xorg/xserver/-/blob/master/hw/xfree86/drivers/modesetting/driver.c):

```c
static Bool
CreateScreenResources(ScreenPtr pScreen)
{
<...>
    err = drmModeDirtyFB(ms->fd, ms->drmmode.fb_id, NULL, 0);

    if (err != -EINVAL && err != -ENOSYS) {
        ms->damage = DamageCreate(NULL, NULL, DamageReportNone, TRUE,
                                  pScreen, rootPixmap);

        if (ms->damage) {
            DamageRegister(&rootPixmap->drawable, ms->damage);
            ms->dirty_enabled = TRUE;
            xf86DrvMsg(pScrn->scrnIndex, X_INFO, "Damage tracking initialized\n");
        }

```

So it looks like damage goes through here:

```c
static int
dispatch_dirty_region(ScrnInfoPtr scrn,
                      PixmapPtr pixmap, DamagePtr damage, int fb_id)
{
    modesettingPtr ms = modesettingPTR(scrn);
    RegionPtr dirty = DamageRegion(damage);
    unsigned num_cliprects = REGION_NUM_RECTS(dirty);
    int ret = 0;

    if (num_cliprects) {
        drmModeClip *clip = xallocarray(num_cliprects, sizeof(drmModeClip));
        BoxPtr rect = REGION_RECTS(dirty);
        int i;

        if (!clip)
            return -ENOMEM;

        /* XXX no need for copy? */
        for (i = 0; i < num_cliprects; i++, rect++) {
            clip[i].x1 = rect->x1;
            clip[i].y1 = rect->y1;
            clip[i].x2 = rect->x2;
            clip[i].y2 = rect->y2;
        }

        /* TODO query connector property to see if this is needed */
        ret = drmModeDirtyFB(ms->fd, fb_id, clip, num_cliprects);

        /* if we're swamping it with work, try one at a time */
        if (ret == -EINVAL) {
            for (i = 0; i < num_cliprects; i++) {
                if ((ret = drmModeDirtyFB(ms->fd, fb_id, &clip[i], 1)) < 0)
                    break;
            }
        }

        free(clip);
        DamageEmpty(damage);
    }
    return ret;
}
```

drmModeDirtyFB() is from [libdrm](https://gitlab.freedesktop.org/mesa/drm/-/blob/master/xf86drmMode.c):

```c
drm_public int drmModeDirtyFB(int fd, uint32_t bufferId,
		   drmModeClipPtr clips, uint32_t num_clips)
{
	struct drm_mode_fb_dirty_cmd dirty;

	memclear(dirty);
	dirty.fb_id = bufferId;
	dirty.clips_ptr = VOID2U64(clips);
	dirty.num_clips = num_clips;

	return DRM_IOCTL(fd, DRM_IOCTL_MODE_DIRTYFB, &dirty);
}
```

This ioctl enters DRM in [drm_mode_dirtyfb_ioctl()](https://elixir.bootlin.com/linux/latest/C/ident/drm_mode_dirtyfb_ioctl) and goes through to [mipi_dbi_fb_dirty()](https://elixir.bootlin.com/linux/latest/C/ident/mipi_dbi_fb_dirty) which does the actual flushing.

```c
static void mipi_dbi_fb_dirty(struct drm_framebuffer *fb, struct drm_rect *rect)
{
<...>
	DRM_DEBUG_KMS("Flushing [FB:%d] " DRM_RECT_FMT "\n", fb->base.id, DRM_RECT_ARG(rect));
<...>
	mipi_dbi_set_window_address(dbidev, rect->x1, rect->x2 - 1, rect->y1,
				    rect->y2 - 1);

	ret = mipi_dbi_command_buf(dbi, MIPI_DCS_WRITE_MEMORY_START, tr,
				   width * height * 2);
```

Enable that logline to see what the DRM driver does flushing wise. See https://github.com/notro/tinydrm/wiki/Development#debug

But this doesn't really help you much since the problem lies somewhere else in the X server.

fbtft is different in that it detects when the framebuffer contents changes and does a flush when that happens, there's no explicit damage tracking.

You could force X to use the the DRM driver fbdev emulation to work aorund the issue, but at the cost of some performance since the buffer is copied in the emulation layer before it's passed on to the DRM driver.

The file /usr/share/X11/xorg.conf.d/99-fbturbo.conf does that on Raspbian in some configurations:
```
Section "Device"
        Identifier      "Allwinner A10/A13 FBDEV"
        Driver          "fbturbo"
        Option          "fbdev" "/dev/fb1"

        Option          "SwapbuffersWait" "true"
EndSection
```

closed 2021-07-07 by notro