notro/tinydrm #28: ST7789 interfacing with stm32mpu running openstlinux 5.4

daniyalyasin93, 2021-06-09:
I am trying to run crystalfonts lcd cfaf240320a0024sc (https://www.crystalfontz.com/product/cfaf240320a0024sc-240x320-full-color-touchscreen-tft-2-4) with stm32mpu over SPI. The LCD controller is ST7789

I am using stmp32mp157 that comes with octavo OSD32MP1-BRK.

Openstllinux kernel version is 5.4

I have build and installed the driver. The driver is loaded correctly. dmesg displays the following:

```
[  109.037910] st7789vw: loading out-of-tree module taints kernel.
[  109.045346] [drm] Initialized ST7789VW 1.0.0 20171128 for spi2.0 on minor 0
[  109.495708] Console: switching to colour frame buffer device 30x30
[  109.652859] st7789vw spi2.0: fb0: ST7789VWdrmfb frame buffer device
```

But I see no output on the lcd screen, whatever I do.

I have used modetest to test the driver as we:

```
root@stm32mp1:~# LIBGL_DEBUG=verbose modetest -M "ST7789VW" -s 31:240x240@RG16
drmOpenDevice: node name is /dev/dri/card0
drmOpenDevice: open result is 3, (OK)
drmGetBusid returned ''
setting mode 240x240-0Hz@RG16 on connectors 31, crtc 34
failed to set gamma: Function not implemented
```

It doesn't display any pattern.

On testing the speed of display:

```
root@stm32mp1:~# LIBGL_DEBUG=verbose modetest -M "ST7789VW" -s 31:240x240@RG16 -v
drmOpenDevice: node name is /dev/dri/card0
drmOpenDevice: open result is 3, (OK)
drmGetBusid returned ''
setting mode 240x240-0Hz@RG16 on connectors 31, crtc 34
failed to set gamma: Function not implemented
freq: 10.32Hz
freq: 10.31Hz
freq: 10.32Hz
freq: 10.33Hz

root@stm32mp1:~# for p in /sys/class/drm/*/status; do con=${p%/status}; echo -n "${con#*/card?-}: "; cat $p; done
SPI-1: connected
root@stm32mp1:~# cat /sys/kernel/debug/dri/0/state
plane[32]: plane-0
	crtc=(null)
	fb=0
	crtc-pos=240x240+0+0
	src-pos=240.000000x240.000000+0.000000+0.000000
	rotation=1
	normalized-zpos=0
	color-encoding=ITU-R BT.601 YCbCr
	color-range=YCbCr limited range
crtc[34]: crtc-0
	enable=0
	active=0
	self_refresh_active=0
	planes_changed=1
	mode_changed=1
	active_changed=1
	connectors_changed=1
	color_mgmt_changed=0
	plane_mask=0
	connector_mask=0
	encoder_mask=0
	mode: "": 0 0 0 0 0 0 0 0 0 0 0x0 0x0
connector[31]: SPI-1
	crtc=(null)
	self_refresh_aware=0

root@stm32mp1:~# modetest -M "ST7789VW" 
Encoders:
id	crtc	type	possible crtcs	possible clones	
35	34	none	0x00000001	0x00000000

Connectors:
id	encoder	status		name		size (mm)	modes	encoders
31	35	connected	(null)-1       	20x20		1	35
  modes:
	name refresh (Hz) hdisp hss hse htot vdisp vss vse vtot)
  240x240 0 240 240 240 240 240 240 240 240 1 flags: ; type: preferred, driver
  props:
	1 EDID:
		flags: immutable blob
		blobs:

		value:
	2 DPMS:
		flags: enum
		enums: On=0 Standby=1 Suspend=2 Off=3
		value: 0
	5 link-status:
		flags: enum
		enums: Good=0 Bad=1
		value: 0
	6 non-desktop:
		flags: immutable range
		values: 0 1
		value: 0
	4 TILE:
		flags: immutable blob
		blobs:

		value:
	20 CRTC_ID:
		flags: object
		value: 34

CRTCs:
id	fb	pos	size
34	38	(0,0)	(240x240)
  240x240 0 240 240 240 240 240 240 240 240 1 flags: ; type: preferred, driver
  props:
	22 ACTIVE:
		flags: range
		values: 0 1
		value: 1
	23 MODE_ID:
		flags: blob
		blobs:

		value:
			01000000f000f000f000f0000000f000
			f000f000f00000000000000000000000
			48000000323430783234300000000000
			00000000000000000000000000000000
			00000000
	19 OUT_FENCE_PTR:
		flags: range
		values: 0 18446744073709551615
		value: 0
	24 VRR_ENABLED:
		flags: range
		values: 0 1
		value: 0

Planes:
id	crtc	fb	CRTC x,y	x,y	gamma size	possible crtcs
32	34	38	0,0		0,0	0       	0x00000001
  formats: RG16 XR24
  props:
	8 type:
		flags: immutable enum
		enums: Overlay=0 Primary=1 Cursor=2
		value: 1
	17 FB_ID:
		flags: object
		value: 38
	18 IN_FENCE_FD:
		flags: signed range
		values: -1 2147483647
		value: -1
	20 CRTC_ID:
		flags: object
		value: 34
	13 CRTC_X:
		flags: signed range
		values: -2147483648 2147483647
		value: 0
	14 CRTC_Y:
		flags: signed range
		values: -2147483648 2147483647
		value: 0
	15 CRTC_W:
		flags: range
		values: 0 2147483647
		value: 240
	16 CRTC_H:
		flags: range
		values: 0 2147483647
		value: 240
	9 SRC_X:
		flags: range
		values: 0 4294967295
		value: 0
	10 SRC_Y:
		flags: range
		values: 0 4294967295
```

I can not make any progress, any help is welcome

notro, 2021-06-11:
Each panel has it's own initialization sequence. The one in st7789vw.c matches a 240x240 panel so the resolution is also wrong. The easy solution is to change jd_t18003_t01_mode to match yours and change jd_t18003_t01_pipe_enable() to match the init sequence in the Arduino SPI Demo code on the product page you linked to.

daniyalyasin93, 2021-06-14:
Thanks for the hint, I have fixed the init sequence. It still doesn't work, I am debugging and will post here details shortly. Anyways I have a question:

https://github.com/notro/tinydrm/blob/d5079aebeb2e2f76036c45bf9a12629f50429788/st7789vw.c#L178

specifies spi mode as mode 3.

But see page 56: file:///home/sedeniustpc08/Downloads/ST7789V_v1.6%20(1).pdf

https://www.newhavendisplay.com/appnotes/datasheets/LCDs/ST7789V.pdf

It clearly says that it reads on rising edge of SCL i.e. CPHA = 0

Also if you see timing diagrams in the datasheet such as on pg 44 it clearly shows that CPOL = 0

Then any good reason for using mode 3?

notro, 2021-06-14:
I don't know why SPI mode 3 is used in the driver, maybe @EliasKotlyar can answer (SPI mode should really be set using DT properties). Looking at the demo code for the [Waveshare display](https://www.waveshare.com/wiki/1.3inch_OLED_HAT) shows that it's using mode 0. 
Have you tried removing that line?

daniyalyasin93, 2021-07-07:
For anyone else who is stuck on this:

please remove alternative pin configuration for sleep mode. The octavo sends the pins to sleep by setting them to Analog mode and they are no longer in SPI mode, hence the drivers don't work. When you remove that they start working

closed 2021-07-07 by daniyalyasin93